Ultimaker/Cura #14507: Z SEAM

padelosGR, 2023-02-07:
### Application Version

5.1

### Platform

w10

### Printer

cr6 se

### Reproduction steps

ALL

### Actual results

PLZ fix the Z seam,  its tragedy

### Expected results

PLZ fix the Z seam,  its tragedy

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

PLZ fix the Z seam,  its tragedy

MariMakes, 2023-02-10:
HEy @padelosGR,

Welcome to the Ultimaker Cura Github 🚀 
Sorry, to hear you have so much trouble with the zseam 😞 

We've made some significant improvements to the Zseam here https://github.com/Ultimaker/CuraEngine/pull/1762

Keep an eye out for the next 5.3 Beta release to try them out. 
I'll be closing this issue since there is a fix in place. 

P.S. Can you please use the issue template correctly next time? Or at least provide a picture of a screenshot of your problem so I, and our lovely moderator volunteers, can troubleshoot your problem? Thank youuuu 🙏

closed 2023-02-10 by MariMakes

padelosGR, 2023-02-10:
Yessss ok, tnx for everything, in all team, keep the good work 🧡