Ultimaker/Cura #14508: Cura Interferes with Other Devices Using a USB UART

timali, 2023-02-07:
### Application Version

5.2.2

### Platform

Windows 10

### Printer

Ender 3 v2

### Reproduction steps

1. Connect some devices (not 3D printers) to your PC via USB UART adapters.
2. Start Cura.

### Actual results

1. Cura opens the UART for these devices, likely because it is searching for 3D printers.
2. Cura interferes with these devices, causing them to become non-functional, and possibly damaging them.

Examples of devices which are affected:
1) 8-channel relay board. When Cura is opened, the relays open and close erratically, causing the attached device to fail.
2) Debugging interface of emergency rescue tool
3) Dragonboard 410c USB bootloader
4) WDC65c816 and WDC65C02 development boards
5) Custom IoT device in development

### Expected results

Likely, Cura is looking for a particular USB/UART chip, and it seems any device which uses this is mistaken for a 3D printer. Running Cura on a machine with these devices plugged in is dangerous because it may cause the devices to malfunction. If Cura cannot reliably differentiate between an actual 3D printer and some other device which is using a USB UART, then there needs to be a way to disable automatic detection of 3D printers so that Cura no longer opens these UARTs and interferes with the attached devices.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

A setting to disable automatic detection (and thus automatic UART opening) should resolve this issue.

GregValiant, 2023-02-07:
Thanks for the report.
The USB Printing plugin is installed by default.  You can un-install it by going to the Cura MarketPlace (the button is at the top right of the screen) and then clicking on the Manage Packages "gear" icon.  Scroll down and disable USB Printing.
Let us know if that works for you.

timali, 2023-02-10:
Disabling the USB Printing plugin prevents Cura from hijacking my other devices connected via USB UART. Thank you for the effective workaround for my issue!

This works for me because I am not currently using USB printing, but for those that are, it would be nice to be able to somehow use USB printing, and not have Cura interfere with other devices. Maybe have the ability to disable automatic USB UART port scanning, and manually specify the UART for your printer(s)?

GregValiant, 2023-02-10:
The problem is that Ultimaker foots the bill for a lot of the Cura development and none of their printers use USB printing as it was deemed a dead end and too hard to maintain (too many different hardware combinations out there).
I looked at it since if someone could get it to really work they would be a hero.  It's not going to be me pulling that particular sword from the stone.  It's noteworthy that no one else has tried either.

I'll go ahead and remove the bug label and close this.  Good luck.

closed 2023-02-10 by GregValiant