Ultimaker/Cura #14511: Combing regression from Cura 4 to Cura 5 when printing the benchy

sidamos, 2023-02-08:
### Application Version

5.2.1

### Platform

Windows 10

### Printer

Creality Ender 2 Pro

### Reproduction steps

This is a regression compared to Cura 4.13.1.

Combing settings are default and the same for Cura 4 and 5. Only a few other settings are modified compared to the bundled Ender 3 profile.

When printing the benchy with those settings in Cura 4, it is perfect. With Cura 5, I get printed lines in the air on the front of the benchy (see photo). 

I can workaround this issue by changing this setting in Cura 5:
Combing Mode “Not in Skin” -> “Not on Outer Surface”

But this is not necessary in Cura 4.

### Actual results

The benchy has some lines in the air with Cura 5.

### Expected results

Print with Cura 5 should look the same as with Cura 4 with the same settings.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

Photo of the issue, sliced with Cura 5:
[image]

Cura 5 project:
[CE3E3V2_3DBenchy-Cura5-Bug.zip](https://github.com/Ultimaker/Cura/files/10685339/CE3E3V2_3DBenchy-Cura5-Bug.zip)

Cura 4 gcode:
[CE3_3DBenchy_C4_ZS.zip](https://github.com/Ultimaker/Cura/files/10685346/CE3_3DBenchy_C4_ZS.zip)

Cura 5 gcode:
[CE3E3V2_3DBenchy_C5_ZS.zip](https://github.com/Ultimaker/Cura/files/10685349/CE3E3V2_3DBenchy_C5_ZS.zip)

MariMakes, 2023-02-10:
Hey @sidamos,

Welcome to the Ultimaker Cura Github 🚀 
Thanks for your report 👍

I always forget which combing mode does what. So I was [reading through the setting in the settings guide ](https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/travel/retraction_combing.md) and it seems to avoid the yellow parts. 
[image]
On the left you see 4.13 and on the right you see 5.2. Because of the variable line width the yellow skin parts have been replaced by the green walls. 

I believe the combing setting you need is Only Within Infill. The dark blue lines are unretracted moves, and the lighter blue lines are retracted moves. You can see more retracted moves in the Within Infill setting. 
[image]
[image]

I'll ask the team to take a look at the Not In Skin combing mode, because that's the only one that acts a bit odd in this model. 
Fingers crossed 🤞

sidamos, 2023-02-10:
Thank you for the quick reaction!

MariMakes, 2023-02-17:
Hey @sidamos,

I discussed your issue with the team.
We've added a ticket to the backlog with the intent to improve this. 
For internal reference CURA-10308

Thanks for the report! 👍