Ultimaker/Cura #14513: Slicing Often Fails in 5.2

bonds0097, 2023-02-08:
### Application Version

5.2.2

### Platform

Windows 11

### Printer

Ultimaker S5

### Reproduction steps

1. Load the attached STL.
2. Slice w/ default Visual Fine settings.

### Actual results

Slicing hangs forever at around 50%

### Expected results

Slicing should complete. 
[UMS5_Halfling_House_4_GF_Unsliced_repaired.zip](https://github.com/Ultimaker/Cura/files/10686313/UMS5_Halfling_House_4_GF_Unsliced_repaired.zip)
[cura.log](https://github.com/Ultimaker/Cura/files/10686321/cura.log)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

No further info.

GregValiant, 2023-02-08:
Thanks for the report and the files.
Someone from the Cura team will take a look.  I do have a comment though.

The "really slow" problem seems to come up a lot with these types of "building" models.  "Pebbled" floors and high definition wood floors that have lots of graining (like your model) create a lot of triangles to go through for the slicer and they are all within a couple of layers of each other.  Your 0.1 Layer Height is causing a lot of calculating in order to figure out the best path for all the "islands" that develop as the floor is printed.
I'm not trying to make excuses and I don't know for sure but it's what I think is going on.
While this is slicing I'm going fishing.  Maybe it will be done when I get back.  It's been 25 minutes and it's only up to the second "c" in Cancel.

GregValiant, 2023-02-08:
Update.
I'm home 4 hours later and the slice indicator in Cura is right where it was.  I was able to Cancel the slice and the program is responding normally.  It just won't slice your model.
If it's any consolation PrusaSlicer hung at 20%.  It finally finished slicing in about 6 minutes and reported that there were missing layers between 45.05 and 45.65 in the Z and there were toolpaths outside the part.  So PrusaSlicer couldn't really slice it either.
Back to MS 3D Builder.  It indicated your repaired model had errors (that is not unusual) so I let it re-repair the model.  Here is a view in Cura.  This is the area on top and to the left of the round hole in the front.  You can see all those little pieces sticking out.  They are a problem.  There are 4 or 6 areas like that around the model.
[image]