Ultimaker/Cura #14516: Multiply Selected hangs opens blank dialog box and hangs Cura (5.2.1)

DukePantarei, 2023-02-08:
### Application Version

5.2.1

### Platform

Windows 11

### Printer

Elegoo Neptune X

### Reproduction steps

1. Open Cura
2. Open a .stl
3. Select the model
4. Ctrl+M or Right click to select Multiply Selected
[image]

### Actual results

The dialog box for multiply items will appear blank and will hang there and will not respond.

[image]

### Expected results

The dialog would populate selections for multiplying the selected model.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/10688475/cura.log)

fieldOfView, 2023-02-08:
Do you have multiple displays? If so, this probably happens with more dialogs (eg the preferences, about window, update profile dialogs, etc). It might help to keep Cura on the primary display.

DukePantarei, 2023-02-09:
> Do you have multiple displays? If so, this probably happens with more dialogs (eg the preferences, about window, update profile dialogs, etc). It might help to keep Cura on the primary display.

I do have multiple displays. Hmm, I was able to get around the issue by editing the .ini files so that the window didn't remember its location on startup and that resolved the hanging on start issue. Not sure if there is a similar setting for dialog boxes. But Cura is on the primary (display0) monitor.