Ultimaker/Cura #14530: bug

Er1kunas, 2023-02-10:
### Application Version

5.2.2

### Platform

windows 10

### Printer

Ender 3 pro

### Reproduction steps

1. Opened model stl
2. Tried to slice and it sliced
3. scaled it up
4. tried to slice again
5. could not slice

### Actual results

could not slice model

### Expected results

sliced up model

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/10704563/cura.log)
model project file is too big for github to upload

GregValiant, 2023-02-10:
Thanks for the report.
I'm not good at reading cura.log files but I sure know this line: "Backend exited abnormally with return code 3221225477!".

I'll mark this as a duplicate and someone from the Cura team will take a look.
A project file would be really useful.  If you could upload the project (or a zip folder containing the project) to a service like Google Drive and provide a download link that would help.

MariMakes, 2023-02-23:
Hey @Er1kunas,

Thanks for your report 👍

Return code 3221225477, is actually a super generic error, even though it doesn't sound generic 😅 
In the Cura 5.3 Beta, we made a fix that should solve a number of these errors. 
Can you download the Beta and retry your slice? 
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.3.0-beta.1

If that's it still doesn't work, we'll need a little bit more information from you.
Do you have a project file for us? It contains the printer and settings we need for troubleshooting. 
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.

closed 2023-03-10 by github-actions[bot]