Ultimaker/Cura #14531: unable to slice

Rosinstein, 2023-02-10:
### Application Version

5.2.1

### Platform

windows 11 pro

### Printer

Creality cr10s

### Reproduction steps

Simply sliced.

### Actual results

asked for me to report the bug slicing failed

### Expected results

4 day gcode

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[CCR10S_Jay 300c v2 Front.zip](https://github.com/Ultimaker/Cura/files/10705500/CCR10S_Jay.300c.v2.Front.zip)
[cura.log](https://github.com/Ultimaker/Cura/files/10705529/cura.log)

GregValiant, 2023-02-10:
Thanks for the report.
Car bodies are tough and the models often have errors.  When I tried to slice the model these warnings came up.
[image]

Your project shows a "Line Width" of 0.4, a "Minimum Line Width" of 0.34mm, a "Support Interface Density" of 10%, and a Support Interface Line Distance" of 0.2mm.  It's that last one that is the problem as it is below your "Minimum Line Width".  If you click on the "fx" button next to "Support Interface Line Distance" it will re-link it to "Support Interface Density".
The errors in the model appear to be repairable.  I used MS 3D Builder to fix it.

This is the repaired model.  The only setting I changed was re-linking the Interface Line Distance as I described above.
[image]

I don't see a bug here but rather a model that needs repair and a settings issue.  Consequently I'm inclined to remove the bug label and close this.  I'll hold off and you can respond here and let me know if you think that is premature.

Rosinstein, 2023-02-10:
Thank you for your prompt response in the past I had no problems with
exports from tinker cad, after sending the bug report I moved the file into
MS 3d Builder, repaired , exported, sliced ok.  Thank you very much!  Funny
we used the same work around.

GregValiant, 2023-02-10:
What I find funny is that people think that programming a robot to make complex shapes is going to be easy.
I like MS 3D Builder.  It isn't very intuitive so some things need to be done several times to get it right.  But it does get it right and when altering STL's it always leave a manifold model.
I'll go ahead and close this then.  Good luck with your project.

closed 2023-02-10 by GregValiant