Ultimaker/Cura #14534: Top/Bottom -> "Line Directions" function doesn't work V5.5.2

SRAsolution, 2023-02-10:
### Application Version

5.5.2 

### Platform

Window 11

### Printer

Own build

### Reproduction steps

1, Import STL
2, rotation 90° (for biggest surface touch plate)
3, Select default profile then "slice"
4, view 1st layer (check line orientation)
5, Change "Top/Bottom Line Direction" value "90" or "60" or "45" or [90] or [60]......
6. press slice and check 1st layer
7. Issue: no change on the layer orientation

### Actual results

New setting has no impact on the build

### Expected results

The lines orientations should change based on the input 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

SRAsolution, 2023-02-10:
> Thanks for the report.
> I can't duplicate the problem. I often use that setting when I rotate a
> model.
> If you would zip a project file and post the zip folder the file may
> provide a clue.
> This is with the "Top/Bottom Line Direction" at [90]
> [image: image]
> <https://user-images.githubusercontent.com/64202104/218088512-b317b5ee-d965-4e04-9f59-8f36fb32c3c5.png>
>
> This is with the setting at [0]
> [image: image]
> <https://user-images.githubusercontent.com/64202104/218089146-81348b1f-a509-4e56-a371-01ac474c952d.png>

Dear Greg,

I attached the files and took the photos.
Note that the 1st appear was on v5.2.1 and I update to 5.2.2 to solve the
problem, but it didn't solve the issue
[image: 0.png]
[image: 90.png]
Kind regards/Üdvötlettel,

*Renato* *Sanchez*
Mobile: +36 20 201 6816
E-mail: ***@***.***

GregValiant, 2023-02-10:
@SRAsolution attachments here don't work via e-mail.  If you could please direct post it here on the website that would work.
(FYI my screenshots were with 5.2.1.)

SRAsolution, 2023-02-11:
[image]
[image]

Re-attached

Since yesterdays I have shut-down my computer and restarted. Now I can't reproduce the issue. (look like it work)
note: before restarting the computer I try to re-lunch  the software, but it didn't solve the issue

closed 2023-02-11 by SRAsolution