Ultimaker/Cura #14536: Slicing failed

d3w4yn3, 2023-02-10:
### Application Version

5.1.0

### Platform

Windows 10

### Printer

Ender 3 Pro

### Reproduction steps

I opened an stl file (attached)
I attempted to slice the file using these settings (attached)
The slice failed with no explanation other than slicing failed.
[Ziro Silk PLA.zip](https://github.com/Ultimaker/Cura/files/10708929/Ziro.Silk.PLA.zip)
[full name plate v 3.0 - 2023-02-06T192125.558.stl.zip](https://github.com/Ultimaker/Cura/files/10708931/full.name.plate.v.3.0.-.2023-02-06T192125.558.stl.zip)
[cura.log](https://github.com/Ultimaker/Cura/files/10709012/cura.log)

### Actual results

"Slicing failed" no other explanation of any kind provided.

### Expected results

Slicing successful

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/10709020/cura.log)

GregValiant, 2023-02-10:
Thanks for the report.
I couldn't duplicate the problem.  The model sliced fine for me.
There is a bug that will keep a good model from slicing when it's at the mid-point of the Cura build plate.  Try moving and/or rotating the model and you may find a spot where it will slice.

As an aside - if you split that model and print the base separately you could print the text with the model laying down.  No "impossible to remove" supports would be required.  2mm holes for location and pieces of filament in them as pins.  Super-glue works very well.
The lettering would be much crisper than trying to deal with the drooping overhangs when the letters are vertical.
[image]

d3w4yn3, 2023-02-11:
I do print it with no overhangs, rotated with letters facing up.

Also, I was able to slice afterward, so it is a rare anomaly.  Thanks for
digging into it!  Still my favorite slicer, by far!!!

GregValiant, 2023-02-12:
I'll go ahead and close this then.

closed 2023-02-12 by GregValiant