Ultimaker/Cura #14540: Crash on high quality

EklDuderino, 2023-02-11:
### Application Version

5.3.0

### Platform

Windows 10

### Printer

Neptune 3 Plus

### Reproduction steps

Increase print speed to 120mm. Increase quality to high, keep changes, slice, program non responsive after clicking slice.

### Actual results

Program non responsive

### Expected results

slicing of file
[cura.log](https://github.com/Ultimaker/Cura/files/10712600/cura.log)
[EN3P_rock_tooth_paste_lid.zip](https://github.com/Ultimaker/Cura/files/10712603/EN3P_rock_tooth_paste_lid.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[EN3P_rock_tooth_paste_lid.zip](https://github.com/Ultimaker/Cura/files/10712604/EN3P_rock_tooth_paste_lid.zip)

GregValiant, 2023-02-12:
Thanks for the report and the file.
That model has some errors and you may need to repair it.  I didn't have any trouble slicing it so I can't duplicate the problem.  There is a bug that can keep models from slicing when they are at the mid-point of the Cura build plate.  Try moving it around and you may find a place where it will slice.  That bug is being worked on but hasn't been fixed yet.
I'll mark this as a duplicate and a slicing error but let us know if that workaround is effective on this model.
[image]

[image]

EklDuderino, 2023-02-12:
Thanks for the response. It wasn't a big deal and I was able to workaround and get this model sliced and printed. I just wanted to send the crash report in hopes it may help. I really enjoy working with Cura. Thanks for the work you do. Cheers.

-------- Original Message --------

jellespijker, 2024-01-14:
Thank you for your report. It seems that your uploaded project file slices correct on our latest unreleased 5.7.0-alpha.0+3abb5 which indicates that the issue you're experiencing seems to be fixed for the upcoming 5.7.0 release.
I'm closing this issue as fixed for the 5.7.0 release
[image]

closed 2024-01-14 by jellespijker