Ultimaker/Cura #14541: Speedbug

Mrakulaus, 2023-02-11:
### Application Version

5.2.2

### Platform

windows10

### Printer

sidewinder X2

### Reproduction steps

under infill setting clicked check box named "connect infill lines"
if this checkbox is pressed i got speed override that can been seen in speed color scheme and also in practice,  if "connect infill lines" unchecked all back to normal.

### Actual results

print speed was overrided, also infill speed is overrided by 65% that can be seen only in the speed color scheme
[image]

### Expected results

[image]

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[curaspeedbug.zip](https://github.com/Ultimaker/Cura/files/10712851/curaspeedbug.zip)
[cura.log](https://github.com/Ultimaker/Cura/files/10712852/cura.log)

GregValiant, 2023-02-12:
Thanks for the report and the files.
I analyzed the gcode with a macro in MS Excel and the infill speed is a consistent 40mm/sec throughout the file.  That's what I see in your settings as well.
[image]

The highest speed in the file is 66mm/sec and is a sprint in the infill on several layers.
This is from the gcode file:
;TYPE:FILL
G1 F300 Z6.95
G1 F1500 E2689.97494                                      
**G1 F3973.5** X155.739 Y103.397 E2690.1306   >this is the 66mm/sec line.  There is one on every layer mid-print.
G1 F1500 E2688.1306
G1 F300 Z7.15
G0 F9000 X154.154 Y104.673 Z7.15
G1 F300 Z6.95
G1 F1500 E2690.1306
**G1 F2400** X154.85 Y104.487 E2690.16804  >the rest of the infill goes down at 40mm/sec
G1 X154.597 Y104.929 E2690.1945
Give your settings this all looks very normal to me and I don't see a bug here at all.  Maybe you could explain further.  In the Cura preview you can see that the infill is going down faster than the walls do (20mm/sec).

Mrakulaus, 2023-02-18:
Hi,
Thank you for your analysis.
As you can see from the pictures the infill speed is set up to 40mm/s where 66mm/sec is coming from? it looks like "connect infill lines" overrides it, also i noticed that other models (another project) dont have this issue.