Ultimaker/Cura #14542: slicing failed

fire51emt, 2023-02-11:
### Application Version

5.2.2

### Platform

windows 10

### Printer

anycubic kobra plus

### Reproduction steps

tried to slice a model.

### Actual results

it says slicing failed

### Expected results

it should slice my model for printing. 

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[CFFFP_r_backTyre_up.zip](https://github.com/Ultimaker/Cura/files/10713499/CFFFP_r_backTyre_up.zip)
[image]

GregValiant, 2023-02-12:
Thanks for the report.
Your model isn't watertight and so Cura can't figure out what is the outside and what is the inside.  I fooled around with it a bit but I couldn't repair it with either Mesh Mixer or MS 3D Builder.
Here is the model in Mesh Mixer.  The magenta balls and surfaces indicate where the errors are.  If you use the Cura x-ray view you will see that those areas coincide with the white high-lighted areas in Cura.  PrusaSlicer wasn't able to slice this either.
[image]

The model has some severe errors so there isn't any way to tell if this is a Cura bug or not.  I hope you understand.
I'll leave this open for a couple of days so you can respond here.

GregValiant, 2023-02-12:
And then I moved the models around and they sliced.
[image]

fire51emt, 2023-02-13:
OK thank you. Sent via the Samsung Galaxy Note20 Ultra 5G, an AT&T 5G smartphone