Ultimaker/Cura #14543: Wipe while retracting

npoulos, 2023-02-11:
### Is your feature request related to a problem?

Just a problem with stringing. Prusa/Slic3r based slicers and S3D all have the ability for the retract move to happen along the wipe move path. Can this be implemented in Cura?

### Describe the solution you'd like

The ability to wipe while retracting.

If the wipe distance is 1mm and the retract is 1mm then the retraction should start retracting at the start of the wipe move, and end retracting once the wipe ends.

### Describe alternatives you've considered

Using other slicers

### Affected users and/or printers

All

InstantMuffin, 2023-03-23:
I would love this to be added as well. This is the only thing I am missing in Cura which single-handedly makes me consider to change slicers.
I love using my eSun PLA+ but regardless of any setting I use (Temp tower, retraction distance tuned, retraction speed as well, travel is high up, z hop disabled, pressure advance (klipper) configured, flow calibration, filament is dry and has been dried as it was unpacked as a precaution to narrow this down, nozzle has been changed/cleaned, printing outside to inside) I still get some residual wisps which nozzle wiping on other slicers immediately fixes.

Shed66, 2023-03-24:
I would love to see this too. I'm struggling to get rid of stringing on an Anykubic Kobra Go and would really like to stick with Cura since that's what I'm used to. Please consider adding this...

DejitaruJin, 2023-06-26:
Someone apparently made [a script](https://www.reddit.com/r/Cura/comments/l7nweh/retraction_before_or_while_wiping/) for this a couple years ago, but it has certain settings requirements that would be less of an issue if it were integrated into Cura. Regardless, we definitely know that it's not a new idea and it is a very impactful feature.

NNMale, 2023-08-10:
Wipe while retracting is a very useful feature. I would like to see this in Cura. Printing with PETG sometimes causes string problems. It would be optimal to pass the nozzle over the inner perimeter and at this time to make a retract. (When printing the outer wall, it returns to the inner wall; when printing infill, the wipe path follows the inner wall) When the nozzle rises from the cooled plastic, there will be no strings.

evranch, 2024-01-01:
Wipe on retract would be great to get into Cura, it's the main reason I sometimes use other slicers despite Cura's far superior pathing and combing behaviour.

Great results on long bowden tube printers compared to coasting or other attempts to compensate, it almost completely removes Z-seam blobbing and gaps especially combined with "retract at layer change". Coasting usually introduces other artifacts while retraction wiping is a more elegant solution.

Rickdiculousme, 2024-12-19:
This sure would be nice to have