Ultimaker/Cura #14549: Slicing Fail

grahamgunderson, 2023-02-12:
### Application Version

5.0.0

### Platform

Windows 10

### Printer

Creality Ender 5 Plus

### Reproduction steps

Just running attached model. Successive layers of same profile aren't uniform; was tweaking endlessly to try getting uniform build on each one, no knobs or weird protrusions. Then this crash.

### Actual results

Slicing failed; asks if I'll submit a bug report. This is it.

### Expected results

Model is vertical columns which should have same pattern each layer. Slicing results in odd knobs on sharp edges, or under-extrusion. Expected consistent gcode for consistent layer geometry.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/10714993/cura.log)
[CE5P_pe22_1 001 stator rod holder half.zip](https://github.com/Ultimaker/Cura/files/10714994/CE5P_pe22_1.001.stator.rod.holder.half.zip)

GregValiant, 2023-02-12:
Thanks for the report and the files.
Your project sliced on the first try for me.  The model doesn't appear to have any errors.  There is a bug that can keep good models from slicing when the model is at the mid-point of the build plate.  Try moving it around in Cura and maybe try rotating it as well.  You may find a place where it will slice.
I'll add the "slicing error" label to this and make it a duplicate report.  I'll leave it open but the Cura team may close this as they have a lot of examples of this particular bug.
[image]

jellespijker, 2024-01-14:
Thank you for your report. It seems that your uploaded project file slices correct on our latest unreleased 5.7.0-alpha.0+3abb5 which indicates that the issue you're experiencing seems to be fixed for the upcoming 5.7.0 release.
I'm closing this issue as fixed for the 5.7.0 release

[image]

closed 2024-01-14 by jellespijker