Ultimaker/Cura #14550: better stl files

bobpombrio, 2023-02-12:
### Is your feature request related to a problem?

It seems that the generated STL files are overly complex and wasted movement on the printer.

### Describe the solution you'd like

I run the STL files thru the free web site NANODLP ,https://www.nanodlp.com/repair-3d-stl-file/ , and this site returns very clean STL code.  Now the end product, the printed STL, looks the same but the printer is spared some unnecessary movements. 

How about offering the option to send the STL thru a "cleaner". Or better yet just run the generated STL thru the cleaner by default.

### Describe alternatives you've considered

There are other sites that provide the same general solution but I feel that NANODLP is the better solution.

### Affected users and/or printers

The printer works smoother and the new STL is much smaller in size.

MariMakes, 2023-02-21:
Hey @bobpombrio,

Welcome to the Ultimaker Cura Github 🚀 
Thanks for sharing that website, I will definitely give it a go. 👍 

Cura already offers some mesh simplification tools like the following settings;
[image] 
Increasing your maximum resolution will already simplify your model. 

Are you aware of the Meshtools Plugin? It can also help with faulty meshes. 
You can get it here: https://marketplace.ultimaker.com/app/cura/plugins/fieldofview/MeshTools

I'll bring it up with the team. But I expect that it will not get a high priority to implement. Fingers crossed 🤞 

Since Cura is open source, you or anyone else is more than welcome to work on adding functionality like you are describing as a plug-in for Cura.

MariMakes, 2023-03-13:
Hey @bobpombrio,

I brought your issue up with the team. 
It's a really ambitious project which is not really a priority of the Cura team since Cura is not built to alter meshes in this way, that's what CAD software is for. 

Since Cura is open source, you or anyone else is more than welcome to work on this issue and create a plug-in yourself. 

To be able to focus on the most requested and needed features and bug fixes from our community and from Ultimaker, we have decided to defer this issue.  We believe we will not work on this anytime soon and will therefore defer/close this issue.

I hope you understand 🙏

bobpombrio, 2023-03-13:
Ok I understand.Thank youSent from my iPad

closed 2024-03-27 by github-actions[bot]