Ultimaker/Cura #14554: i have a problem when printing i get so far and then it retracks my filament backwards untill all the way out

Gavin1972, 2023-02-13:
### Application Version

5.2.1

### Platform

windows 11

### Printer

ender 3 v2

### Reproduction steps

i sliced my print and saved it then started the print so far though it started springing the filament backwards until there was no filament left in the tube. 

### Actual results

i had to re start all over again

### Expected results

it shouldn't of happened to start with

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

iv now gone to prusa slicer. untill this problem is fixed

GregValiant, 2023-02-13:
Thanks for the report.
Without a project file we can't tell what's going on.  Consider though that hundreds of thousands of people use Cura without this problem so I'm pretty sure it's a settings issue.
Load a model and set Cura up to slice.  Use the "File | Save Project" command to create a 3mf file.  Zip the file and post the zip folder here.  My guess is that somehow you have "Absolute" extrusion and "Relative" extrusion mixed up in your StartUp Gcode.

Asterchades, 2023-02-14:
I've heard of this sort of thing happening in the past with poorly configured post-processing. Typically the result of a script either configured to or requiring relative extrusion mode during an absolute mode print, resulting in it using either a very low or even negative E value. When you're a decent chunk through the print this can effectively perform hundreds of millimetres of retraction, easily feeding back past the extruder.

Though as mentioned the project file will be necessary to establish this, or at the very least the GCode in use. The project will list what if any scripts are in use, where the GCode would hopefully at least have notation as to where said scripts have been applied.

Sineos, 2023-02-15:
Coincidentally I just had this issue last weekend. Was experimenting with the retraction settings and tuned off combing completely (think it was not skin before) and the resulting gcode had exactly this issue.
Not sure if I'm able to reproduce tho.

closed 2023-03-02 by github-actions[bot]