Ultimaker/Cura #14556: Slicing Fails

d-portero, 2023-02-14:
### Application Version

5.2.2

### Platform

Windows 10 

### Printer

Ultimaker S5

### Reproduction steps

Load Crystal Dragon Model. Hit Slice. Slice Fails

### Actual results

Error says slicing failes

### Expected results

Slice should have completed 

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[image]
[slicing error CrystalDragon_NewLegs_Curled_2(1).zip](https://github.com/Ultimaker/Cura/files/10727853/slicing.error.CrystalDragon_NewLegs_Curled_2.1.zip)

GregValiant, 2023-02-14:
Thanks for the report.
Is that a re-make of another Crystal Dragon model?  That model has been in here before.  It had hundreds of internal errors that couldn't be repaired.
This model brought Mesh Mixer to it's knees when I attempted to repair the model.
You can see here that MS 3D Builder doesn't like it either.  MS 3D Builder was able to repair it.
[image]

So after repairing the model and moving it around on the build plate (a known bug) it did slice.

It is hard to tell with this model whether this is Cura bug.  Since I had to move it around to get it to slice it is a definite duplicate of that other bug and I'll mark it as such.  The Cura team has plenty of examples regarding that bug (including another Crystal Dragon) and may close this when they get to it.  Take a look at #12546.  It looks a lot like this dragon.
[image]

d-portero, 2023-02-14:
It is the crystal dragon from cinderwing3d. Cura did was not slicing anything tonight, so I have restated the computer and will see if it works better tomorrow. 

I have only ever sliced the crystal dragon in other slicer, today was the first day in Cura. The other slicer must not have recognized, or ignored, the internal errors. I will try and fix the model. 

I was aware of the rotation bug, but rotating the model did not fix anything. 

Thanks for your help!