Ultimaker/Cura #14558: Dual extrusion fail on the first color change using shared nozzle

Drawork, 2023-02-14:
### Application Version

5.3.0-alpha+xmas

### Platform

Windows 10 x64

### Printer

Lotmaxx SC-10 Shark V2

### Reproduction steps

Using a dual extrusion printer with a shared nozzle and two steppers for each color I've found a bug or something that needs too be fixed because it ruins every printing.
First color: Purple
Second Color: Yellow
These are my settings for both colors:
[image]
[image]
The first color change happens at layer 11 when yellow color needs to be purged on the prime tower and then print the yellow part of the model:
[image]
It seems that everything is ok but checkin the G.code
[image]

PURPLE 1: The printer prints the last line
PUEPLE 2: Retracts 80mm
PURPLE 3: Reset Extruder
YELLOW 4: Printer to changes to second extruder
YELLOW 5: Extruder reset to 0
YELLOW 6: It retracts for traveling instead of inserting the second color
YELLOW 7 AND 8: Printer starts the second color

### Actual results

The first color change happens at layer 11 when the yellow color needs to be purged on the prime tower and then print the yellow part of the model:
[image]
It seems that everything is ok but checking the G.code
[image]

PURPLE 1: The printer prints the last line
PURPLE 2: Retracts 80mm
PURPLE 3: Reset Extruder
YELLOW 4: Printer to changes to the second extruder
YELLOW 5: Extruder reset to 0
YELLOW 6: It retracts for traveling instead of inserting the second color
YELLOW 7 AND 8: The printer starts the second color

G1 F2700 X119.893 Y115.368 E667.23205
G1 F3600 E587.23205
G1 F300 Z2.4
G92 E0
**T1**
G92 E0
**G1 F1800 E-5**
;MESH:chips.stl
G0 F7200 X119.893 Y115.368 Z2.4
G1 F300 Z2.2
G0 F7200 X71.003 Y41.295
G0 X66.3 Y39.12
;TYPE:PRIME-TOWER
G1 F1800 E0
G1 F3600 X67.445 Y41.262 E0.08078
G1 X68.149 Y43.584 E0.16148

And as result, we get this gap:
[image]

### Expected results

There is nowhere for the printer to insert the second filament to start printing and just changing the GCODE on these lines
[image]
Changing the retraction -5 to 80 to insert the second color and then reset the extruder to 0 with G92 E0 fix all the problems, but there is no way to implement this initial insertion, it seems that Cura interprets that the second filament is already inserted.

G1 F2700 X119.893 Y115.368 E667.23205
G1 F3600 E587.23205
G1 F300 Z2.4
G92 E0
**T1**
G92 E0
**G1 F1800 E80
G92 E0**
;MESH:chips.stl
G0 F7200 X119.893 Y115.368 Z2.4
G1 F300 Z2.2
G0 F7200 X71.003 Y41.295
G0 X66.3 Y39.12
;TYPE:PRIME-TOWER
G1 F1800 E0
G1 F3600 X67.445 Y41.262 E0.08078
G1 X68.149 Y43.584 E0.16148

The final result
[image]

Thank you and best regards 😁

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

G1 F[feed] E[mm to inject]
G92 E0

3Duke3, 2023-02-14:
issue #10092 was a different problem to yours. 

I am unfamiliar with a "2 in 1 out" setup as you have described, but if you install the "Printer Settings" Plugin, there is an "Extruders Share Nozzle/Heater" settting, that I imagine will have some relevance to your issue.

[image]

[image]

GregValiant, 2023-02-15:
Thanks for the report.
"Shared Nozzle Initial Retraction" is the setting and as @3Duke3 says, it's in the Printer Settings.
This snippet is from a 5.2.1 slice without the Initial Retraction.  (I've turned on Relative Extrusion so it's easier to read.)
G1 F4500 X46.637 Y143.652 E0.00674
G1 F3000 **E-80**
G1 F600 Z0.7
T1
(...)
G1 F1500 **E-5**
M204 S500
(...)
;TYPE:SKIRT
G1 F1500 **E5**

This is a slice with Initial Retraction set to 80mm
G1 F4500 X46.637 Y143.652 E0.00674
G1 F3000 **E-80**
G1 F600 Z0.7
T1
(...)
G1 F1500 **E75**
M204 S500
(...)
M205 X10 Y10
;TYPE:SKIRT
G1 F1500 **E5**

So I don't see a bug as it is a settings issue.  @Drawork you will need to set Cura up to match your printer's configuration.
I'll leave this open though so you can respond here.  Let us know if it is resolved.

Drawork, 2023-02-15:
OK, thank you very much, I really appreciate it, I had already activated both comments on the .cfg file

machine_extruders_share_heater = True
machine_extruders_share_nozzle = True

GregValiant, 2023-02-15:
Nice.  Thanks for getting back.
I'll go ahead and close this then.

closed 2023-02-15 by GregValiant