Ultimaker/Cura #14559: Assignment of numbers when objects are duplicated and the order in which they are displayed in the object list

Km0107, 2023-02-14:
### Application Version

5.2.1

### Platform

Windows10

### Printer

Custom FFF

### Reproduction steps

I duplicated the same object to 200 pieces.

### Actual results

As a result, replication can be done successfully.
However, the display in the object list is
"xxx.stl,xxx.stl(1),xxx.stl(10),xxx.stl(100)..."
which is very inconvenient when counting.

[image]

### Expected results

The display in the object list should be 
"xxx.stl,xxx.stl(1),xxx.stl(2),xxx.stl(3)..."
Otherwise counting the number of identical objects would become very tedious.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

This problem does not adversely affect the operation of the software, but it does adversely affect the user's emotional state.
At least to me.

MariMakes, 2023-02-15:
Hey @Km0107,

Welcome to the Ultimaker Cura Github 🚀 

Oh yea, this reminds me of renaming a bunch of files or folders in Windows. They seemed to have resolved it somehow because I cannot reproduce it to show you 😅 

I'll bring it up to the team to see what they can do to improve it. But I expect that if it's not a quick fix that it will not get a high priority to implement. Fingers crossed 🤞 

Can you share why it's important to you to count the items on the buildplate after multiplying?

Km0107, 2023-02-15:
I am a printing agent for a 3D printer as a side business.
It is a small but crucial task to see which and how many pieces of data will be printed in order to perfectly fulfill the customer's request.

> 

I am a printing agent for a 3D printer as a side business.
It is a small but crucial task to see which and how many pieces of data will be printed in order to perfectly fulfill the customer's request.

Km0107, 2023-02-15:
> Hey @Km0107,
>
> Welcome to the Ultimaker Cura Github 🚀
>
> Oh yea, this reminds me of renaming a bunch of files or folders in Windows. They seemed to have resolved it somehow because I cannot reproduce [...]

I am sorry.
I closed it by mistake.
I am a beginner and am not familiar with how to use it.
Are you able to reply with this?

I am a printing agent for a 3D printer as a side business.
It is a small but crucial task to see which and how many pieces of data will be printed in order to perfectly fulfill the customer's request.

Km0107, 2023-02-15:
Thank you.
I would like something like that, but it is not a good idea to have it printed as a physical item.
It would add to the waste.
Can't we use a support blocker to handle this?

MariMakes, 2023-02-15:
The plugin is free to download, so you can give it a go yourself.  💪

Km0107, 2023-02-15:
I will wait for the team's response as I try it out.
Thanks for the useful plugin introduction.

5axes, 2023-02-21:
Not sure the [NameIt](https://marketplace.ultimaker.com/app/cura/plugins/5axes/NameIt) plugin could solved your issue. But the last release on the [Github](https://github.com/5axes/NameIt/releases) could provid a workaround. By using the Rename integrated function , the name in this case are correctly displayed in the right order.  

Original Order 
[image]

Rename NameIT function
[image]

Km0107, 2023-02-21:
That's exactly what I was looking for!
Thank you!

MariMakes, 2023-05-15:
Hey @Km0107,

Sorry, I totally forgot to update you on what happened with the developer team!

They agreed that it's valuable to whow how many files there are selected,
We've added a ticket to the backlog with the intent to improve this. 
For internal reference CURA-10305

Thanks for the report! 👍