Ultimaker/Cura #14561: CR6-MAX on Cura...?

Fabdu57, 2023-02-14:
### Is your feature request related to a problem?

I have CR6-MAX but this printer do'nt exist on Cura...

### Describe the solution you'd like

Is somebody have ressources files for this printer ?
(Or configuration ?)

### Describe alternatives you've considered

Is somebody have ressources files for this printer ?
(Or configuration ?)

### Affected users and/or printers

CR6-MAX

### Additional information & file uploads

Is somebody have ressources files for this printer ?
(Or configuration ?)

GregValiant, 2023-02-15:
The printer definitions are submitted to Ultimaker by members of the community or by the manufacturers.  The question has been asked about that printer before in request #11551.  Creality actually seems to be pretty good about supplying definition files.

MariMakes, 2023-02-21:
Hey @Fabdu57,

I believe the printer is available in the [Creality Slicer](https://www.creality.com/pages/download-cr-6-max?spm=..page_1934481.products_display_1.1) but not yet in Cura. 

Since Cura is open source, you or anyone else is more than welcome to work on adding this 3d printer and create a pull request yourself. You can find instructions on how to add a new 3D printer definition here: https://github.com/Ultimaker/Cura/wiki/Adding-new-machine-profiles-to-Cura 

If you are stuck in any way, please reach out! I'll be happy to help out.

closed 2023-02-21 by MariMakes