Ultimaker/Cura #14562: Print settings don't honor filament settings on filament change.

N6REJ, 2023-02-15:
### Application Version

5.3.0-alpha-xmas

### Platform

win 10x64 pro

### Printer

Aquila S2

### Reproduction steps

Change filament type and then inspect print settings like "retraction"

### Actual results

They will not match

### Expected results

When filament is chosen it overwrites print settings at first.

What I expect to happen is...
you have your print settings from the last time you selected them.
you change your filament so you load settings that are required for that filaments
print settings change.
IF you then change print settings then those changes are used. 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

doesn't matter what filament or printer profile is used

GregValiant, 2023-02-15:
Thanks for the report.
There is a hierarchy involved.  If you have a stock profile like "Draft" or "Standard" active then yes, the material will override the profile settings.  But if you have a custom settings profile active then that takes precedence and the material will not override it.  At least that's my understanding of how it's supposed to work.  I find it easier to keep custom setting profiles for the different materials and tweak them for the specific model that I'm printing.  So in my case the materials never override my profile because that's how I've set it up.
So what you have here is confusing but expected behavior.  I'm going to remove the bug label but I'll leave it open for now.  Let us know if you have additional thoughts or feel that closing it would be premature.

N6REJ, 2023-02-15:
like you say thats very confusing.  Someone turned me onto a plugin that lets me r-click in the box and it will load material settings but idk if I have to do that for each item or if it changes them all.
Perhaps a switch could existing in settings to allow filament to overwrite custom.

GregValiant, 2023-02-15:
The Material Settings plugin and the Printer Settings plugin were written by the same person (@fieldOfView).  Cura works in a world of plugins but those two are kind of crucial to what goes on.  Why they are plugins and not part of the main program I suppose is an internal decision at UM but it's the way it is.
I'll go ahead and close this then.  Good luck.

closed 2023-02-15 by GregValiant