Ultimaker/Cura #14564: Management of files, settings and profiles

FoxtrotLimaOscar, 2023-02-15:
### Is your feature request related to a problem?

Assume you created several Cura profiles for different needs along with suitable printer and material settings. Let's say you used one of these profiles called "functional" to create a G-Code and print it. For later use, you save the Cura project as .3mf file.

**Several months or even years later:**
In the meantime you fine-tuned your "functional" profile, your printer settings (for example Start and End G-Code) and the material settings.
Now let's say you want to reprint your old project. So you load your old .3mf project file "as project".
It gives you the option to update existing printer, profile and material settings or to create new settings from the file.

- I don't want to overwrite everything I optimized in the last year. But with one accidental click on "open" this is exactly what would happen
- I don't want to create a completely new profile then called "functional#1", "functional#2" and so on. This clutters everything.

Why is there only the option to create a new profile (then containing the old settings) or to overwirte an existing optimized profile (again with old settings)? In my eyes this doens't reflect the usecase of a save / load functionallity.

### Describe the solution you'd like

If you open an old .3mf Cura project to your current Cura setup, it should be possible to load the printer settings, the profile settings and the material settings contained in this old .3mf-file while neither overwriting the current profile nor creating a new profile or changing the current printer and material setup.

After the old project file was loaded, its stored old print settings **should be shown as temporary changes from the current profile without acually changing the current profile**. It should further be selectable to use the current printer and material setup instead of the setup from the old file. **Adding two options each in the drop down menu appearing when opening a file should allow this selection and solve the problem:**

**- "Load as temporary change to current settings"**
**- "Use current settings**

_(An example is shown below in "additional information" section.)_

Temporary changes to the currenct settings from the loaded file should be marked with the reset arrow to allow going back to the current setting's default value.
The choice "update existing" and "create new" should be removed imho. If desired, the print settings can still actively be saved as new profile afterwards or overwrite an existing profile. The "load as temporary change" setting should be the standard choice to avoid overwriting existing profiles with old ones.

I know, there is a global option to only load the model. But often I want to use the current pritner and material settings and still load the profile settings from an old file as changes to the existing profile.

With additive manufacturing being an important part of industry, research institutions and private persons nowadays, it should be expected, that an iterative optimization process will be performed on settings by all users. Why would anyone want to overwrite that effort with old settings with no going back?

### Describe alternatives you've considered

Switch to another slicer tool. Cura's handling of settings loaded from old files is soooo annoying, that I'm willing to give up on all the other Cura benefits.

### Affected users and/or printers

All

### Additional information & file uploads

Illustration of the feature request:
[image]

Making use of the already existing Cura feature highlighting temporary changes to the current profile:
[image]

GoguyT3D, 2023-02-18:
Yes I have tried to read/understand this thoroughly and I completely agree. For me the problem arises when I load an old 3mf for my same printer,filament but I only want the "updated since last save" options to appear, and if I've changed the name of the profile I'm phuked because it pastes the name of the old not supported profile, crashes when I click it and can't be removed(without serious digging in python definition files all carefully linked like a big spiderweb ready to burst with a "your configuration seems to be corrupt ") What I wanted and need is the ability to select from a dropdown list of "Changed settings not similar to 3mf" and individually select what I want updated for a supported profile, no added profile, and my experience would be perfect.

5axes, 2023-03-09:
Duplicate : https://github.com/Ultimaker/Cura/issues/13223 ?

FoxtrotLimaOscar, 2023-03-09:
[#13223](https://github.com/Ultimaker/Cura/issues/13223) is quite similar to my request but not as specific in describing the problem and the desired / suggested solution. Most importantly, the suggested solution deviates from my request: Especially my desired "Open File(s)"-menu option to load a project file with the actual settings from the loaded project **marked as change to the current profile** instead of overwriting and existing or adding a new profile is not described in #13223. The "keep current" option alone as suggeted in #13223 would not satisfy my request. Therefore, i would not rate my request as duplicate - nevertheless it partly contains [#13223](https://github.com/Ultimaker/Cura/issues/13223).

stopanko, 2023-05-22:
Fully **AGREE**. Have spent days changing, Printing, changing.. Opened Old Project **ONCE**. And have lost all my work.
That's a disaster.

Should be prioritized. 

Why don't transfer `Project` settings if differ as `YOUR`(by hand) unsaved changes instead of overriding all my work.
I have lost my tuning couple of times now.  

SHOULD BE PRIORITIZED!!!.

Marskramer, 2023-07-10:
In my opion there shouldn't be any options at all when you open a project.
When you open a project, 99 out of 100 times you don't remember the settings exactly. How can you ever diced if you want to do something with a profile that you can't even glance over?
Profile's or template file settings should never be connected to the saving or opening process of a project file. It should be totally separated.

When you open a project, it should simply open with all the settings that it was saved with. Then if you want to update it to a newer profile, you load this profile from a database. If you want to save a profile, you save it to the database. Simple and intuitive. This way project files can also work as some sort of backup database for  profiles.  All CAD software works like this for example.

Things will stay messy or tedious in some situations non the less, because so many settings are saved in just 2 (printer and material) profiles. But that's an other topic I guess. But overall it would be an improvement. Material profiles would be an solid and secure basis to start with for each new project. Something that has to be and also safely can be tweaked. Instate of an insecure "hope this was a good one" situation.

I've had several prints with a burned top, because of an old end g-code bug that keeps coming back through opening project files that "update" (more like downdate) my printer profile in a whim.
Workaround for now is to open the project and let Cura make new profiles. Then load old profiles and delete new unwanted profiles . Tedious.
OR alternatively start from scratch with the new profiles. Even more tedious when a project consist of multiple .stl objects, tweaked supports etc.

GoguyT3D, 2023-07-10:
> In my opion there shouldn't be any options at all when you open a project. When you open a project, 99 out of 100 times you don't remember the settings exactly. How can you ever diced if you want to do something with a profile that you can't even glance over? Profile's or template file settings should never be connected to the saving or opening process of a project file. It should always be separate processes/tasks.
> 
> When you open a project, it should simply open with all the settings that it was saved with. Then if you want to update it to a newer profile, you load this profile from a database after opening the project. If you want to save a profile, you save it to the database. Simple and intuitive. This way project files can also work as some sort of backup database for profiles. All CAD software works like this for example.
> 
> Things will stay messy or tedious in some situations non the less, because so many settings are saved in just 2 (printer and material) profiles. But that's an other topic I guess. But overall it would be an improvement. Material profiles would be an solid and secure basis to start with for each new project. Something that has to be and also safely can be tweaked and copied to old projects. Instead of an insecure "hope this was a good one" and "crap, I lost the good one" situation.
> 
> I've had several prints with a burned top, because of an old end g-code bug that keeps coming back through opening project files that "update" (more like downdate) my printer profile in a whim. Workaround for now is to open the project and let Cura make new profiles. Then load the correct printer profile and delete the new unwanted profile. Tedious. OR import models only. Tedious, as I first have to fetch the old material profile from the project, taking care I don't overide anything.

I am no coder so it probably sounds more simple than it is to implement,  but what you are currently describing sound to me like you would much rather "import models" more often than not because that's what I have to do without my previous comment coming back to haunt me. For me I am always "tweaking" my profiles so no/any 3mf doesn't bring up the updated settings I have recently saved per profile. Read my previous comment and you'll see were in the same boat.

Marskramer, 2023-07-10:
> I am no coder so it probably sounds more simple than it is to implement, but what you are currently describing sound to me like you would much rather "import models" more often than not because [...]

That works in some situations, but the process feels awkward and doesn't work in half of the situation.

For example, I'm someone that doesn't print very frequently. Maybe once a month. What often happens when I want to print something new is that I remembered an old print that was similar to the new one and that went well. The old project was weeks ago, so I have no idea what profile I used back then. Easy way would be to find the old project that I cataloged nicely on my PC, open the .3mf and swap the 3D shape for the new one. Currently Cura forces me to make new profiles every time I want to do this. No other way to open the old project with the profiles. Resulting in a huge list of profiles of which you have no clue what to pick for a new project later on.

Different example. This month I'm actually working on many projects, consisting of many parts. Maybe one project is an assembly of 5 parts printed with the same material and quality. Only all 5 parts are in seperate .3mf files and have different bed adhesion, support types and infill settings. If I want to be able to open those projects several times, maybe because the 3D shape changed after printing the first prototype. I can no do this quickly. I either have to save a new profile in Cura for every .3mf ( with very specific settings that I never gone use on an other project) or override profiles with a risk of unwanted overrides of a more basic profile.

Printer profile even more so. Rarely change anything there, only got one printer. Currently Cura still forces me to be very careful when opening older projects if I ever happen to change anything.

I'm no coder ether, but doesn't seem hard at all to code my proposal. Settings are currently already saved in both the .3mf and in Cura. Just a matter of disabling the popup that shows up when you open a .3mf and instead load the settings form the .3mf without overriding any profiles that are saved in Cura. Probably dumping the settings from the .3mf in a temp file behind the scenes.

batesman1, 2023-12-19:
I want this feature SO BADLY. I often go back to an old project file where I had the print settings dialed in. I HATE that it overwrites my machine settings and custom G-Code when I open it as a project. I don't understand why the machine settings cannot be decoupled from the project OR give the user the option to import them as well as the print settings. 

I've had a number of prints fail when I re-printed an old project because I'm now using ABL, but wasn't in my old machine settings and the start G-Code didn't use the ABL.

suttonb, 2024-03-07:
This has been a constant source of frustration for me as well. When opening a project file the program can clearly tell which settings have overrides, why can't these be loaded as temporary overrides to an existing profile instead of changing the profile stored on the disk?

nbering, 2024-07-14:
I agree. Some print jobs can be very finicky, requiring very specific support or adhesion setups, that deviate from my normal settings. I'd rather store than in a project with the geometry, than as a custom profile.

I'm very unlikely to use those settings for another project, but if I want to load it up to slice for settings with another brand of filament or something, I don't want to re-do all of the configuration from scratch.

At the moment, I just ignore that projects even exist, doing all my setup over again. I take notes for the tricky projects, on paper, or in the CAD document where that's supported.

nickweedon, 2024-10-22:
+1 I find this very annoying it would be great if i could load a project and have the option to just keep my current settings.