Ultimaker/Cura #14568: Slicing error

GusThings, 2023-02-15:
### Application Version

5.2.1

### Platform

Window 10

### Printer

Ender 3

### Reproduction steps

.a

### Actual results

Error on slicing slt 
[Index_Finger.zip](https://github.com/Ultimaker/Cura/files/10743175/Index_Finger.zip)

### Expected results

.a

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

.abc

GregValiant, 2023-02-15:
Thanks for the report and the file.
As you can see below I didn't have any trouble slicing it.  The model doesn't appear to have any errors.
There is a bug in Cura (it's being worked on) that can be fooled by moving and/or rotating the model on the Cura build plate.  Give that a try and let us know if it works.
I'll mark this as a duplicate.  After you respond I'll close this as the Cura team has a lot of examples to work with.
[image]

jellespijker, 2024-01-14:
Thank you for your report. It seems that your uploaded project file slices correct on our latest unreleased 5.7.0-alpha.0+3abb5 which indicates that the issue you're experiencing seems to be fixed for the upcoming 5.7.0 release.
I'm closing this issue as fixed for the 5.7.0 release

[image]

closed 2024-01-14 by jellespijker