Ultimaker/Cura #14571: Slicing Failed

madmaxh3, 2023-02-15:
### Application Version

5.2.1

### Platform

Windows 10

### Printer

Creality Ender 3 S1-Pro/with Sonic Pad

### Reproduction steps

Loaded "Wednesday Thing.stl" into Cura.
Clicked on slice.

### Actual results

Slicing failed

### Expected results

sliced

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

Project file too large to upload.
[cura.log](https://github.com/Ultimaker/Cura/files/10745007/cura.log)

GregValiant, 2023-02-15:
Thanks for the report.
This is in the log file **"Backend exited abnormally with return code 3221225477!"**
I'm not sure what it means but I've seen it a lot here.  I'll make this a duplicate and someone from the Cura team will take a look.

MariMakes, 2023-02-23:
Hey @madmaxh3 ,

Thanks for your report 👍

Return code 3221225477, is actually a super generic error, even though it doesn't sound generic 😅 
In the Cura 5.3 Beta, we made a fix that should solve a number of these errors. 
Can you download the Beta and retry your slice? 
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.3.0-beta.1

If that's it still doesn't work, we'll need a little bit more information from you.
Do you have a project file for us? It contains the printer and settings we need for troubleshooting. 
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.

closed 2023-03-10 by github-actions[bot]