Ultimaker/Cura #14574: The 3D printer does not output enough material

THESAMISM, 2023-02-15:
### Application Version

5.2.2

### Platform

Windows 11

### Printer

Creality Ender-3 pro

### Reproduction steps

1. I printed and the printer does not output enough meterial.
2. I checked if there is problem with the printer (for example: nozzle too close to brint bed, nlocked nozzle, stripped filament) and everything was fine.
3. After that i checked id there was a problem with the settings in Cura (for example: filament diameter, density, temperature, etc) and there was no problem either.

### Actual results

The print remains the same.

### Expected results

The printer was supposed to print normal.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

.

GregValiant, 2023-02-15:
Thanks for the report.
Your not giving us much to go on.  You can zip a gcode file and upload the zip folder here.
Check your printer.  There should be a setting "Control / Filament" make sure that Volumetric Extrusion is turned off.

THESAMISM, 2023-02-15:
It does not support that file type. I checked and Volumetric Extrusion is turned off. Which more information do you need to understnd the problem?

GregValiant, 2023-02-16:
Load a model in Cura.  Set Cura up to slice.
Select the "File | Save Project" command to create a 3mf file.  That file will contain your printer, all your settings, and the model.
Go to Windows Explorer and right click on the 3mf file.  Select "Send To..." and then "Compressed (zip) Folder".
Find one of your gcode files that has the problem and drag-and-drop it into that zip folder.
Post the zip folder here.

THESAMISM, 2023-02-16:
[xyz-Calibration-Cube-Castle.zip](https://github.com/Ultimaker/Cura/files/10753130/xyz-Calibration-Cube-Castle.zip)

GregValiant, 2023-02-16:
I used a macro and MS Excel to analyze your gcode.  The macro calculates the Line Width from the amount of material extruded.  At .20 layer height it shows a consistent Line Width of 0.40mm throughout the file.  Those are the settings you used when the gcode was created.  That means that the flow output is exactly correct in the gcode.
[image]

My printer is also an Ender 3 Pro.  The Creality Ender series are notorious for developing partial blockages where the bowden tube meets the back of the nozzle.  I would suggest taking the hot end apart and cleaning it out.
Have you calibrated the E-steps on the printer?  Most folks end up at about 97 and that's where mine is.
A second possibility is that there is a problem with the E-stepper chip on the mainboard.  I had one expire and had to replace the mainboard.  A symptom of an E-stepper driver chip going bye-bye is under-extrusion.

This is your gcode printed on my E3Pro.  It looks good to me.
[image]
[image]

I don't see a bug here but rather a printer problem.  I'll go ahead and remove the bug label and close this.  I hope you understand.  Whatever you have going on there is no problem in the Cura generated gcode.

closed 2023-02-16 by GregValiant