Ultimaker/Cura #14579: 5.2.1 slicing failed

mikey46656, 2023-02-16:
### Application Version

5.2.1

### Platform

windows 11

### Printer

ender 5

### Reproduction steps

Attempted to slice a model for printing.

### Actual results

Error message: 'slicing failed. slicing failed with an unexpected error. please consider reporting a bug on our issue tracker.'

### Expected results

should have sliced the model

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

I zipped the file but I can't upload it because it is 37MB, so this is useless.

Asterchades, 2023-02-17:
Without a project file, or at least the base STL or 3MF in question, little if any diagnostics can be done.

Per the instructions when creating the issue, if the file is too large to upload to GitHub then you'll need to use a 3rd party hosting service. The one mentioned by name is WeTransfer, but any such service is fine - eg GDrive, MediaFire, Mega.

mikey46656, 2023-02-20:
My apologies - my brain was not working properly. I'm genuinely sorry about
this, and the issue was easily fixed.

-m

MariMakes, 2023-02-21:
Hey @mikey46656,

Do I understand correctly that your issue has been resolved? 
Or did you attempt to share your projectfile?

To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.

closed 2023-03-07 by github-actions[bot]