Ultimaker/Cura #14589: Cura 5.2.2 has bug slicing my STL file from TinkerCAD

TrinhHuynh08, 2023-02-17:
### Application Version

5.2.2

### Platform

Window 10

### Printer

Creality Ender 3 V2

### Reproduction steps

I tried to slice a stl file I made in TinkerCad but it says slicing error

[image]

### Actual results

NA
I dont see any step to follow

### Expected results

NA

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

Asterchades, 2023-02-17:
Without a project file, or at least the STL in question, little if any diagnostics can be done.

Note that files cannot be attached via the e-mail response system. They must be uploaded through the website. Changing the file extension of the project from .3MF to .ZIP will allow it to be uploaded.

TrinhHuynh08, 2023-02-17:
Hi,

Sorry I am new to Github and 3D printer. I will make a change with more
detail. Sorry for this inconvenience.

Trinh

GregValiant, 2023-02-17:
If you go to the Microsoft App Store you can download and install MS 3D Builder.  Open your model in 3D Builder (you may also have to also use the "insert" button) and if there are errors it will find them and you can repair them.  Surfaces facing inside instead of outside are the biggest problem and they are actually an easy fix.
If you don't find any errors then try moving and/or rotating the model on the Cura build plate.  There is a known bug regarding slicing errors when the model is at the build plate mid-point.

MariMakes, 2023-02-21:
Hey @TrinhHuynh08,

Welcome to the Ultimaker Cura Github 🚀 
That's a pretty cool TinkerCad model 💪 

Are you still stuck or has your issue been resolved?

If that's not it, I'll need a little bit more information from you.
Do you have a project file for us? It contains the printer and settings we need for troubleshooting. 
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.

TrinhHuynh08, 2023-02-22:
[M_letter.zip](https://github.com/Ultimaker/Cura/files/10799779/M_letter.zip)
Hi everyone @Asterchades @MariMakes @GregValiant ,
My apology for late response. First of all, thank you for all of your responses to support newbie like me. Attached in all files I have made for this little project.

- File name bottom_print-copy.stl is the orginal .STL file I export from TinkerCad, this file was the one had error when I tried to slice it in Cura 5.2.2. I then followed @GregValiant advice to download MS 3D Builder to fix it. Although I wished MS 3D Builder told me what was the error or mistake I made, so that I can improve next time, after fixing, I was able to slice it in Cura. 
I grouped two parts (the base 2mm and the wall outline 2mm, the wall stayed on the top of the base (Z-axis of the wall was +2mm)). After slice it in Cura, in the first print layer in my 3D printer (ender 3 V2) it was unconnected line. I then make file named base

- File name base: this file in tinkercad, instead of having the wall stay on the top of the base, I had the wall get inside of the base (Z-axis of the wall was 0mm), sliced it in Cura. When I printed out, it printed out the base perfectly but when it printed the wall, the wall thickness was printed out only 1mm instead of 2mm. And on the top right of the M, the wall position leaned more to the right which result it off the base about 3mm. Half of M letter on the left side printed out perfectly, the base and the wall merged perfectly but the wall was till only 1mm thick

-File name Top_print: In tinkercad, I did the same as file name base but after that, I saved it in Tinkercad as my own create shape because I thought it would be better if I can make these two pieces instead of grouping, it become a whole. I then used my own create shape export as .STL, dropped it in Cura, sliced no error. But when it was printed, it was continuous line but not a straight line, it was straight then went half of rectangular then straight. I had to leave my house at that time, so I decided to cancel the print instead of let it finish the print to see if after few layers it will fix it itself or not.

Should I make each piece (the base and the wall) in Tinkercad separately, export them in different files, then merge them later in Cura?

GregValiant, 2023-02-22:
When I measure the wall in MS 3D Builder it appears to be .66mm thick.  That is the dimension that Cura has to work with.
When I view the gcode files in Cura there is no "leaning".

I was going to explain my suggestions but it's easier to provide a project file.  I used the "base" file for this.  Use the "advanced" settings and go through the settings to see what I've done.  The result is a 2mm thick base with 2mm walls.  "Remove all Holes" is turned on so remember to turn it off when you are done.
[GV M_base.zip](https://github.com/Ultimaker/Cura/files/10805572/GV.M_base.zip)

[image]

TrinhHuynh08, 2023-02-22:
Thank you @GregValiant . Let me redo again next weekend to make improvement.

closed 2023-04-21 by github-actions[bot]