Ultimaker/Cura #14596: Cure Ultimaker 5.2.2 fails to slice .3MF object

agtrandicosta24, 2023-02-17:
### Application Version

5.2.2

### Platform

Windows 10

### Printer

AnyCubic Mega Zero

### Reproduction steps

Launch Cura Ultimaker.

Load file from local drive.

Click slice.

### Actual results

Error Message populate after slicer begins.

[image]

### Expected results

File should have sliced and allowed for save to thumb drive.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

https://drive.google.com/file/d/1vCSnzCGL4w9TujCd52Xgeacp2H1OBAuV/view?usp=share_link

GregValiant, 2023-02-18:
Thanks for the report.
I don't have permission to download that file and there isn't much I can do without it.

GregValiant, 2023-02-20:
Ta-Daaa
[image]

I didn't find any errors in the model.  After moving it and rotating it sliced.
This is a well known bug in Cura 5.x.  It's being worked on.  I'll mark this as a duplicate and close it as the Cura Team has a lot of examples of this bug.  I hope you understand.

closed 2023-02-20 by GregValiant