Ultimaker/Cura #14598: Slicing Failed

Corattikid, 2023-02-18:
### Application Version

5.2.1

### Platform

Windows 10

### Printer

CR-10S Pro V2

### Reproduction steps

I added a file to the slicer and clicked "Slice" error continues and slice fails

### Actual results

Slicing failed with unexpected error

### Expected results

Should slice program so i can 3D print it 
[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

When I try to slice the file it continues to fail

MariMakes, 2023-02-21:
Hey @Corattikid,

Welcome to the Ultimaker Cura Github 🚀 

Is it possible that your printjob is located at 0, 0, 0, and it slices if you move it a little away from the center of the buildplate? 
Then it might be a duplicate of https://github.com/Ultimaker/Cura/issues/13643

If that's not it, I'll need a little bit more information from you.
Do you have a project file for us? It contains the printer and settings we need for troubleshooting. 
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.

MariMakes, 2023-02-21:
Oh! I think I recognize your model. 
It might be a duplicate of https://github.com/Ultimaker/Cura/issues/13445
They were successfully able to slice the file in Cura 4.13

closed 2023-03-07 by github-actions[bot]