Ultimaker/Cura #14602: unable to slice

mgdavila, 2023-02-18:
### Application Version

5.2.1

### Platform

Win 11

### Printer

Hellbot Hidra Plus

### Reproduction steps

opens stl file
slice

### Actual results

[image]

### Expected results

[image]

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/10774739/cura.log)

GregValiant, 2023-02-18:
Thanks for the report and the log.
_"Backend exited abnormally with return code 3221225477!"_
I have no idea what that means but it has shown up a lot in log files.  The Cura team will take a look at this.  As for me, I'm thinking about playing that number in the lottery.

MariMakes, 2023-02-23:
Hey @mgdavila,

Thanks for your report 👍

Return code 3221225477, is actually a super generic error, even though it doesn't sound generic 😅 
In the Cura 5.3 Beta, we made a fix that should solve a number of these errors. 
Can you download the Beta and retry your slice? 
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.3.0-beta.1

If that's it still doesn't work, we'll need a little bit more information from you.
Do you have a project file for us? It contains the printer and settings we need for troubleshooting. 
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.

closed 2023-03-10 by github-actions[bot]