Ultimaker/Cura #14608: Failed uploading with creality cloud

solorzanob, 2023-02-19:
### Application Version

4.13

### Platform

windows 10

### Printer

ender 3

### Reproduction steps

1. i sliced the model
2. i try uploading the model to creality cloud
3. model had been compressing 
4. in step 3/4 uploading the model cant been uploading. 'Upload filed' 

### Actual results

i cant uploading the model

### Expected results

upload the model to creality cloud

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[image]

Asterchades, 2023-02-19:
Creality Cloud integration is provided by a plugin, not the base Cura client. As such the issue will need to be dealt with by the plugin and its author.

https://github.com/crealitycloud/CrealityCloudIntegration

closed 2023-02-19 by Asterchades

gueleratsource, 2023-02-19:
I started having the same issue after the update on my MacBook M1. Kinda going crazy since nothing seems to solve this issue...