Ultimaker/Cura #14609: Slicing adds two additional planes which do not exist in the model

andiwinter, 2023-02-19:
### Application Version

5.2.2

### Platform

Windows 10

### Printer

Flsun Super Racer

### Reproduction steps

Do standard slicing.

### Actual results

At layers 43 and 63 an additional plane is inserted, but the model is hollow.

### Expected results

A correct slicing.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[Googles13.zip](https://github.com/Ultimaker/Cura/files/10777049/Googles13.zip)

GregValiant, 2023-02-20:
Thanks for the report.
This is an interesting one.  The bottom line is that the problem is the model.  Now for the rest of the story...
I exported the STL from Cura and opened it in MS 3D Builder.  No errors.  So I opened it in Mesh Mixer.  No errors.  So I uploaded it to [formware.co](https://formware.co/OnlineStlRepair) online stl repair.  No errors.
But wait,...after processing the formware file that had no errors it sliced.  The same thing with the Mesh Mixer file after I used the Inspector tool to repair the "perfect model".  It sliced.  So I'm sure it's the model, but I have no idea why.

Here is the original in the center with the Mesh Mixer file on one side and the formware one on the other.
[image]

I'm satisfied this is a model problem so I'll go ahead and close this.  I hope you understand.  I'll attach the unrepaired/repaired model that sliced.  I'll play with this some more. as I'm curious about what's going on.

[FSR_Goggles13_fixed.zip](https://github.com/Ultimaker/Cura/files/10778516/FSR_Goggles13_fixed.zip)

closed 2023-02-20 by GregValiant

andiwinter, 2023-02-20:
Many thanks for your effort and explanations. I've never had an issue like this with Cura, and was wondering why it failed especially with such a simple model.

andiwinter, 2023-02-20:
@GregValiant I found the issue - you gave the hint with formware.co online repair. I tried it out and the log says: 256 inverted faces. So I inverted the 128 vertical faces on the inside of the model with MeshLab, and voila Cura is able to slice the model. I already was wondering why MeshLab showed the inside faces so dark - the reason was that the normals have been in the wrong direction. 
[image]

GregValiant, 2023-02-20:
Nice work and you probably learned something as well.  When I ran it through the formware repair site it showed no errors, but when sliced it acted like there were flipped normals.  That's why I kept trying different tools.