Ultimaker/Cura #14615: Infill not working properly

Scottz1000, 2023-02-19:
### Application Version

5.2.1

### Platform

Windows 11

### Printer

Ender 3 

### Reproduction steps

I tried different infill patterns.

This happened on 5.2.1, 5.2.2, and 5.3.0

### Actual results

It printed just like the screen shot I added below. Infill did not go all the way out to the outer walls
[image]

### Expected results

To correct the problem, I can't be the only one who noticed it

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

GregValiant, 2023-02-20:
Thanks for the report.
Actually, as far as I know you are the first to have this problem.  I can't duplicate it.
With that model loaded and Cura set up to slice use the "File | Save Project" command.  Zip the 3mf file and post the zip folder here.  It could be a bug.  The project file will tell.

Scottz1000, 2023-02-20:
[Egg_arrowhead Custom PLA lw 0.4mm lh 0.2mm if 10% ext1 210C bed 55C.zip](https://github.com/Ultimaker/Cura/files/10778562/Egg_arrowhead.Custom.PLA.lw.0.4mm.lh.0.2mm.if.10.ext1.210C.bed.55C.zip)

GregValiant, 2023-02-20:
Uncheck "Infill as Support".  When it's checked the infill acts like support and only builds a tower to support the overhang of the roof.
I'll go ahead and close this as a settings issue.  I hope you understand.

closed 2023-02-20 by GregValiant