Ultimaker/Cura #14618: "Unable to slice" error for seemingly no reason, goes away after changing *any* variable

username43734, 2023-02-20:
### Application Version

5.2.1

### Platform

Windows 10

### Printer

Two Trees Sapphire Plus

### Reproduction steps

1. Insert 3d model file
2. Adjust printer settings
3. Try to slice
4. Get "Unable to slice" error
5. Change *any* variable
6. Change that variable back to what it was
7. Slice successfully

### Actual results

(Only happens occasionally) Cura shows "Unable to slice" error

15 second video that demonstrates the issue:
https://youtu.be/-A9OF2HBZJU

### Expected results

Slice successfully

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/10778385/cura.log)

MariMakes, 2023-02-20:
Hey @username43734,

Welcome to the Ultimaker Cura Github 🚀 
Wow! It looks like you changed a lot of settings in your project file. 😮 

Is it possible that your printjob is located at 0, 0, 0, and it slices if you move it a little away from the center of the buildplate? 
Then it might be a duplicate of https://github.com/Ultimaker/Cura/issues/13643

If that's not it, I'll need a little bit more information from you.
Do you have a project file for us? It contains the printer and settings we need for troubleshooting. 
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.

MariMakes, 2023-02-21:
Hey @username43734,

Thanks for getting back to us. 
We cannot troubleshoot your issue if there are no solid steps to reproduce. 
Even if we made a fix we would not be able to check if the fix would work. 

I'll be closing this issue, I hope you understand. 
If you have solid steps to reproduce, please feel free to create a new issue 🙏

closed 2023-02-21 by MariMakes