Ultimaker/Cura #14622: Just updated and z-seam still not aligned

16lionheart16, 2023-02-20:
### Application Version

5.2.2

### Platform

Windows 10

### Printer

ender 3 v2

### Reproduction steps

imported model that had a cylinder.
used normal settings, with z seam set to user-specified & corner preference set to none, sliced.
z seam is not aligned, tried different positions and corner preference, no difference.
tried z seam relative, no difference
tried updating from 5.1.0, no difference
[image]

### Actual results

z seam all over the joint
[image]

### Expected results

z seam to be a straight line along cylinder

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[Z Seam.zip](https://github.com/Ultimaker/Cura/files/10788241/Z.Seam.zip)

MariMakes, 2023-02-21:
Hey @16lionheart16,

Welcome to the Ultimaker Cura Github 🚀 
Thanks for your report 👍

We've made some significant improvements to the Zseam here https://github.com/Ultimaker/CuraEngine/pull/1762
Keep an eye out for the next 5.3 Beta release to try them out, your model will look like this: 
[image]

I'll be closing this issue since there is a fix in place.

closed 2023-02-21 by MariMakes

16lionheart16, 2023-02-21:
> Ultimaker/CuraEngine#1762

Hi, would it be possible to get the gcode for that, or the project file,
thanks

MariMakes, 2023-02-21:
Sure! Here you go:

[53_CE3E3V2_REELPART_V1.zip](https://github.com/Ultimaker/Cura/files/10794250/53_CE3E3V2_REELPART_V1.zip)