Ultimaker/Cura #14625: Ender3 Neo Supported

YuHao-Wang-EL3MENT, 2023-02-21:
### Is your feature request related to a problem?

There is no neo 3 in the printer, please add it, THANKS!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!

### Describe the solution you'd like

There is no neo 3 in the printer, please add it, THANKS!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!

### Describe alternatives you've considered

There is no neo 3 in the printer, please add it, THANKS!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!

### Affected users and/or printers

There is no neo 3 in the printer, please add it, THANKS!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!

### Additional information & file uploads

There is no neo 3 in the printer, please add it, THANKS!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!

Asterchades, 2023-02-21:
Printer definitions for non-Ultimaker printers are provided by the community - not the Cura team.

Good news is that a pull request already exists from the community to add this particular printer, and its successor (?). So chances are it will be implemented in the next release (5.3) if not shortly after. If you have anything to contribute I would encourage you to drop by the discussion here: https://github.com/Ultimaker/Cura/pull/13580

In the meantime you can still use the printer. Printer definitions don't contain anything absolutely required to run any given printer, so you can adapt an existing profile by simply updating settings such as the build volume, print head dimensions, and Start GCode.

I'll be closing this for now.

closed 2023-02-21 by Asterchades

tonylambiris, 2023-02-25:
Did you see the comment I made yesterday about plate dimensions?
https://github.com/Ultimaker/Cura/pull/13580#issuecomment-1444013077