Ultimaker/Cura #14626: Cura is not able to slice Skull

reite24, 2023-02-21:
### Application Version

5.2.2

### Platform

Windows 10

### Printer

Anycubic Vyper

### Reproduction steps

I designed a dartholder at Tinkercad, by using a skull as base and adding 12 Holes. Then i sliced it and printed it. Because the holes were to small, i modified the diameter of the holes with Tinkercad. After that I tried to slice the Skull v2, but I got an error massage.

### Actual results

Slicing the Skull with Cura is not possible.

### Expected results

The Skull shoud be sliced.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

STL Data of Skull v2:
https://cloud.sturm-gruppe.com/index.php/s/SdPrbcyPtFWJ8dH

Error massage:
[image]

MariMakes, 2023-02-21:
Hey @reite24,

Welcome to the Ultimaker Cura Github 🚀 
Thanks for your report 👍

[image]

This mesh seems faulty
Have you tried doing a quick mesh fix with the Meshtools Plugin? 
You can get it here: https://marketplace.ultimaker.com/app/cura/plugins/fieldofview/MeshTools
If that doesn't fix it you can try a more elaborate mesh fixer like https://myminifactory.github.io/Fast-Quadric-Mesh-Simplification/.
Or you can use the "Make Solid" feature in AutoDesk MeshMixer. https://meshmixer.com/download.html

reite24, 2023-02-21:
Thank's for the quick reply. Meanwhile i could slice the Skull. I just moved it a bit in x and y direction and then it worked.

MariMakes, 2023-02-24:
Yay! Good to know, I'll close this ticket.

closed 2023-02-24 by MariMakes