Ultimaker/Cura #14631: Parameter "index" is not declared.

picarica, 2023-02-21:
### Application Version

5.2.2

### Platform

Linux Gentoo

### Printer

Creality cr-10

### Reproduction steps

trying to open fila via browse files (the file button on top left) or via file> open files , or saving to disk doesnt open file exporer (in this case nemo)

### Actual results

nothing opened and just got error message

### Expected results

open nemo file exporer

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

error from terminal is 
`Fontconfig warning: "/etc/fonts/conf.avail/05-reset-dirs-sample.conf", line 6: unknown element "reset-dirs"
Cyclic dependency detected between "file:///tmp/.mount_UltiMaf6ubsl/share/cura/resources/qml/Actions.qml" and "file:///tmp/.mount_UltiMaf6ubsl/share/cura/resources/qml/Actions.qml"
QQmlEngine::setContextForObject(): Object already has a QQmlContext
QFSFileEngine::open: No file name specified
qt.qml.context: file:///tmp/.mount_UltiMaf6ubsl/share/cura/resources/qml/Settings/SettingComboBox.qml:24:9 Parameter "index" is not declared. Injection of parameters into signal handlers is deprecated. Use JavaScript functions with formal parameters instead.
QFSFileEngine::open: No file name specified
`
and logs in question are here
[cura.log](https://github.com/Ultimaker/Cura/files/10796820/cura.log)
[cura.1.log](https://github.com/Ultimaker/Cura/files/10796822/cura.1.log)

MariMakes, 2023-02-23:
Hey @picarica,

Welcome to the Ultimaker Cura Github 🚀 
Sorry to hear you are stuck 😞 

I'm not a Linux user but I've seen more reports of this. 
Could you try the workaround mentioned here: https://github.com/Ultimaker/Cura/issues/11874#issuecomment-1401090909?

picarica, 2023-02-26:
> Hey @picarica,
> 
> Welcome to the Ultimaker Cura Github rocket Sorry to hear you are stuck disappointed
> 
> I'm not a Linux user but I've seen more reports of this. Could you try the workaround mentioned here: [#11874 (comment)](https://github.com/Ultimaker/Cura/issues/11874#issuecomment-1401090909)?

well they recommen to not use modern version but non-modern version doesnt even launch i have many libGL errors 
like
`Could not probe OpenGL. This program requires OpenGL 2.0 or higher. Please check your video card drivers.`
and this is terminal output
```
PyInstaller/loader/pyimod02_importers.py:499: DeprecationWarning: the imp module is deprecated in favour of importlib and slated for removal in Python 3.12; see the module's documentation for alternative uses
Gtk-Message: 11:29:31.717: Failed to load module "xapp-gtk3-module"
Gtk-Message: 11:29:31.719: Failed to load module "canberra-gtk-module"
libGL error: MESA-LOADER: failed to open radeonsi: /tmp/.mount_UltiMajmniEQ/libstdc++.so.6: version `GLIBCXX_3.4.30' not found (required by /usr/lib/llvm/15/lib64/libLLVM-15.so) (search paths /usr/lib64/dri, suffix _dri)
libGL error: failed to load driver: radeonsi
libGL error: MESA-LOADER: failed to open radeonsi: /tmp/.mount_UltiMajmniEQ/libstdc++.so.6: version `GLIBCXX_3.4.30' not found (required by /usr/lib/llvm/15/lib64/libLLVM-15.so) (search paths /usr/lib64/dri, suffix _dri)
libGL error: failed to load driver: radeonsi
libGL error: MESA-LOADER: failed to open swrast: /tmp/.mount_UltiMajmniEQ/libstdc++.so.6: version `GLIBCXX_3.4.30' not found (required by /usr/lib/llvm/15/lib64/libLLVM-15.so) (search paths /usr/lib64/dri, suffix _dri)
libGL error: failed to load driver: swrast
Fontconfig warning: "/etc/fonts/conf.avail/05-reset-dirs-sample.conf", line 6: unknown element "reset-dirs"

```
not sure why it doesnt work i have radeon RX 6500XT card but modern version works fine
so that workaround wont work for me, and is probably a different issue

picarica, 2023-02-26:
nevermind after installing sys-apps/xdg-desktop-portal-gtk on gentoo, this solved the issue, thank you for help, but still non-modern version doesnt work but shouldnt be an issue

closed 2023-02-26 by picarica