Ultimaker/Cura #14632: curaproifle export visibilitiy

Craftcentric, 2023-02-21:
### Application Version

5.0.0 beta

### Platform

windows11

### Printer

cr10 s

### Reproduction steps

tried to export cura profile
then i tried to locate it in the file explorer
it shows up in recent files
 it is not visible outside of that context it recognizes the file creation but cura will not actualize it
 
 

### Actual results

nothing i lose the files
[image]
[image]

### Expected results

i should be able to see them and transfer the files theyre not even visible to cura

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[image]
 how much more info can i provide

MariMakes, 2023-02-23:
Hey @Craftcentric,

Welcome to the Ultimaker Cura Github 🚀 
Thanks for your report 👍

We just released a new Cura 5.3 Beta that makes it easier to save and update your profiles by adding a save button. We introduced a lot of impressive changes since the 5.0 beta a year ago. 
[image]

Can you help me understand why you are exporting and importing profiles? 
Are you trying to keep settings for a certain model and revisit them? 

That might be easier with project files. Those contain the printer and settings you need for tuning. 
To save a project file go to File -> Save project.

Craftcentric, 2023-02-27:
so the main reason is yes to retain some associated settings  together with the designation of that specific printer im trying to give the printer profile to a coworker so we can reliably slice files the same way for this printer so being able to trasnfer profiles would be great! you say i should try exporting the profiles i created for the printer itself? as a project>?

MariMakes, 2023-02-27:
If you are looking for a reliable way to share settings for models with colleagues. 
Especially if you are working on the same printers sharing project files is the easiest way. 

It's what we designed the Digital Library around, and what I use every day. 
[You can read more about it here.](https://support.makerbot.com/s/article/1667398835941)

To me, it doesn't seem like a bug in Ultimaker Cura anymore. 
Would you still like to keep this issue open or do you agree that we can close it?

Craftcentric, 2023-02-27:
well  the thing is the settings arent model specific but as long as i can create a profile this way sure, but then change the term to oversight because thats a very backwards way to share setting considering it says i can export printer profiles and then i cant so no its not resolved 

however i will try your method but if the settings do not continue past the sliced model then yeah still an issue

MariMakes, 2023-05-12:
Hey @Craftcentric,

Sorry, it took us a while to get back to you 😞 
Did you have a chance to try if the digital library is something for you? 

If that's not it perhaps this can help. I learned this week that Cura profiles are not text files but are .zips. So if you want to alter them to share that's something to take into account. You can read more about it here https://github.com/Ultimaker/Cura/issues/15477#issuecomment-1545660227

If that's not what you are looking for, 
I'll need a little bit more information. How can I still help with this issue?

Craftcentric, 2023-05-12:
exporting the slicing profile as a project definitley helped but its clunky and not very intuitive way to ensure another user has the neccesary settings to use a new printer but thanks