Ultimaker/Cura #14633: Slicing failed with an unexpected error

jk101290, 2023-02-21:
### Application Version

5.2.1

### Platform

Linux Mint 20.3

### Printer

Ender Extender 400XL (Modded E3Pro)

### Reproduction steps

Tried slicing a 3D model

### Actual results

Message: Slicing failed with an unexpected error

### Expected results

Cura slices the model

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

This does not occur if I rotate the model so that it is along the y axis. It also does not occur in 5.1.0. Including project and log.

[xtender400XL_profilebug.zip](https://github.com/Ultimaker/Cura/files/10797800/xtender400XL_profilebug.zip)
[cura.log](https://github.com/Ultimaker/Cura/files/10797863/cura.log)

MariMakes, 2023-02-23:
Hey @jk101290,

Thanks for your report 👍

It seems to be a duplicate of https://github.com/Ultimaker/Cura/issues/13643
I was able to slice it with the 5.3 Beta, can you try aswell? 
[image]

You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.3.0-beta.1

jk101290, 2023-02-24:
@MariMakes I was able to slice it in 5.3, thanks!

closed 2023-02-24 by jk101290