Ultimaker/Cura #14635: "Remove Raft Inside Corners" produces extraneous chunks of raft

pjscopeland, 2023-02-22:
### Application Version

5.2.2

### Platform

MacOS Monterey 12.3.1 M1

### Printer

Voxelab Aquila X2

### Reproduction steps

1. `Build Plate Adhesion Type` = `Raft`.
2. `Remove Raft Inside Corners` = `true`.
3. `Raft Extra Margin` = `5.0` mm. (Not sure how relevant this is to _causing_ the problem, but it does affect the shape of the extra chunk.)
4. Allow to slice.

### Actual results

In each layer of the raft there are a couple of triangles that get printed first, and then the entire layer is printed over top of them (so these sections get printed twice).

### Expected results

There should be no extra triangles, just the main raft.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

Top view of layer 1
[image]

Project file: [Archimedeans.3mf.zip](https://github.com/Ultimaker/Cura/files/10799426/Archimedeans.3mf.zip)

GregValiant, 2023-02-23:
Thanks for the report and the file
This appears to be associated to the "Raft Extra Margin" which you have set to 10mm.
Here is layer 1 at 11mm Raft Extra Margin.  You can see that the triangle thing has mostly gone away.  There are a couple of left over marks in there.
[image]

and here it is at 9mm
[image]

So it would appear that something is going on with the way Cura calculates the outer margin of the raft.  It might have something to do with your model layout as well because they appear to be less than "Raft Margin X2" apart and so each raft actually interferes with the next one over.  Looking at it that way it may be a simple settings issue.  Someone from the Cura team will take a look at this.
(After playing with it a bit I thought it looked best at 6mm Raft Margin and moving a couple of those parts away from each other each would be on it's own raft.  I think this would print fine with a brim as well.)
[image]

In regards to your models - Cura has colored those surfaces red because there are errors in those two models.  I was able to repair them with the Cura Mesh Tools plugin.  Errors in models can often lead to oddities in the tool path as Cura can't understand where the outside is and where the inside is.  If Cura can't precisely calculate the model then the raft shape can be off.
[image]

pjscopeland, 2023-02-23:
Thanks heaps @GregValiant. Unfortunately this isn't the only model I'm having this trouble with – but the other one I'm working on at the moment is proprietary, or at least I paid for it, and I don't imagine the artist would appreciate me sharing it publicly, even to fix a bug.

I also prefer bigger rafts as they are harder to dislodge, and brims have never worked great for me. Could be my entry-level printer, of course.

But thank you very much for the Mesh Tools tip. I couldn't figure out what Cura was complaining about – even in Sketchup everything looked fine – but apparently I needed to "Fix model normals". Something must have gone wrong with the mirroring in the source.

MariMakes, 2023-02-23:
Hey @pjscopeland,

Welcome to the Ultimaker Cura Github 🚀 

Increasing the raft extra margin value already seems to help a lot 👍 
I'll bring it up to the team to see what they can do to improve it. 

Fingers crossed 🤞

MariMakes, 2023-03-13:
Hey @pjscopeland,

I've discussed your issue with the team.

Since there is a workaround in place, increasing the raft extra margin value, we expect we will not work on this anytime soon and will therefore defer this issue. 

Since Cura is open source, you or anyone else is more than welcome to work on this issue and create a pull request yourself.

I'll close the issue, I hope you understand.