Ultimaker/Cura #14638: Slicing Error

Drubix, 2023-02-22:
### Application Version

5.2.2

### Platform

Windows 11

### Printer

Artillery Sidewinder X2

### Reproduction steps

1. Loaded Dodge Challenger SRT Demon HPE 1200 from Thingiverse
2. Scaled to 200%
3. Attempted to Slice

### Actual results

Immediate Error message and report bug button

### Expected results

Cura should slice the vehicle for printing

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[stderr.log](https://github.com/Ultimaker/Cura/files/10806881/stderr.log)
[D.Challenger Slice Fail.zip](https://github.com/Ultimaker/Cura/files/10806883/D.Challenger.Slice.Fail.zip)

MariMakes, 2023-02-23:
Hey @Drubix,

Welcome to the Ultimaker Cura Github 🚀 
Thanks for your report 👍

Your Project file seems to be corrupt. 
Can you try uploading it again? It contains the printer and settings we need for troubleshooting. 
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser. 

We've fixed a big cause for the crashes in Cura 5.3, can you try if your file slices in the beta?
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.3.0-beta.1

Drubix, 2023-02-23:
Actually the beta has seemed to work this error out already, thank you for the help!

closed 2023-02-23 by Drubix