Ultimaker/Cura #14640: Crashes and UI issues with Apple M1 and Ventura

Bonwit, 2023-02-22:
### Application Version

5.2.2

### Platform

OSX Ventura

### Printer

Ender 3 V2

### Reproduction steps

Select "orbit" with the right mouse button, then try to exit the orbit by releasing the mouse button.

### Actual results

The orbit becomes stuck, where you are unable to stop it by releasing the mouse button.

### Expected results

You should be able to start and stop the orbit using the right mouse button.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

When orbiting the selection using the right mouse button, sometime the orbit sticks, where releasing the button does not allow allow you to get out of the orbit mode. The only recourse is to quit Cura and restart. Also, at times the program will crash on its own.

MariMakes, 2023-02-23:
Hey @Bonwit,

Welcome to the Ultimaker Cura Github 🚀 

I recognize your issue, there are some cases where I encounter this on Windows as well. 
In my case I can often leave orbit mode by pressing Escape, is that also an option for you?

Or, do you have a Wacom Tablet? Because then it might be a duplicate of https://github.com/Ultimaker/Cura/issues/12554.

Bonwit, 2023-02-23:
Thanks! I’ll give escape a try next time it happens. Best,

MariMakes, 2023-06-05:
Hey @Bonwit,

We are cleaning up our open issues, so we can focus on the most requested and needed features and bug fixes from our community and from UltiMaker, I stumbled upon your issue about orbiting. 

Did you encounter the issue again?
Did pressing escape resolve this for you?

We are learning more and more about the crashes. You can follow our investigation here: https://github.com/Ultimaker/Cura/issues/15161

Would you still like to keep this issue open or do you agree that we can close it?

Bonwit, 2023-06-05:
Not sure what you mean by “orbit”, but it crashes all the time, usually while idling in the background. M1 Mac running Ventura.

MariMakes, 2023-06-06:
Hey @Bonwit,

Thanks for the reply. 😊 
This seems to be a duplicate of https://github.com/Ultimaker/Cura/issues/15161
We have a ticket on our backlog with the intent to improve this behavior.

I'll be closing this issue as a duplicate but you can follow the progress here: https://github.com/Ultimaker/Cura/issues/15161

closed 2023-06-06 by MariMakes