Ultimaker/Cura #14641: Highlight Unconnected Overhangs

SilverBead, 2023-02-22:
### Is your feature request related to a problem?

It's frustrating to think you don't need supports only to find that a small section hangs low and would be printing in thin air if there were no supports. An unconnected overhang always demands extra attention and should be highlighted. 

### Describe the solution you'd like

Have a way to highlight overhangs that would be floating in thin air without supports. By this I mean a section of the model that has nothing below it and nothing to the side of it to attach to. 

### Describe alternatives you've considered

I have tried slicing the whole model with supports and then checking each layer to find unconnected overhangs. But, this is laborious. Especially with complex character models. 

### Affected users and/or printers

This could happen for anyone but especially those printing complex character models.

Asterchades, 2023-02-23:
Overhangs - including bridges - are already coloured in red on the Prepare view, based on the configured Support Overhang Angle (which unfortunately cannot be altered without enabling Supports). How does this differ from your proposal?

SilverBead, 2023-02-23:
On some intricate character models, there are many tiny overhangs. Many can be safely ignored since they are small enough to print without supports. But, if unconnected overhangs are not caught, a printer can start outputting filament over just air ...causing the whole model to fail. In the attached picture, there are many dozens of overhangs. If all were connected to the model, I might be able to print it with only supports on the hands. But in truth, it will fail in some key spots. 

[image]