Ultimaker/Cura #14643: New BETA not slicing print.

TheAgentHamster, 2023-02-22:
### Application Version

5.3.0

### Platform

Windows 11

### Printer

Custom

### Reproduction steps

Just clicked slice with my desired settings and it came up with an error. "Slicing Failed With an Unexpected error".

### Actual results

"Slicing Failed With an Unexpected error".

### Expected results

Nothing happens

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[Star_trek_Enterprise_NCC-1701_1_2000_2138859.zip](https://github.com/Ultimaker/Cura/files/10809188/Star_trek_Enterprise_NCC-1701_1_2000_2138859.zip)
I was trying to slice the original full enterprise. The fork/stand sliced but the full model wouldn't.

GregValiant, 2023-02-23:
Thanks for the report.
Congratulations.  I do believe you have the first bug report for the new beta version.
I can't duplicate this as it sliced for me on the first try.  Someone from the Cura team will take a look.

TheAgentHamster, 2023-02-23:
> Hey @TheAgentHamster <https://github.com/TheAgentHamster>,
>
> Welcome to the Ultimaker Cura Github 🚀
>
>    - Congratulations on having one of the first bugs in the Beta 🏆
>
> The STLs slices with my desired settings:
> [image: image]
> <https://user-images.githubusercontent.com/40423138/220920086-fe7a8627-7565-4d20-95fe-d788e64ee24b.png>
>
> Do you have a project file for us? It contains the printer and settings we
> need for troubleshooting.
> To save a project file go to File -> Save project. Please make sure to
> .zip your project file. For big files, you may need to use WeTransfer or
> similar file-sharing sites.
>
> You'll need to open Github in your browser to attach any files to your
> ticket
> You can click the *view it on GitHub* link at the bottom of your email to
> open GitHub in the browser.

I will make sure to get that to you in a few hours when I get chance.

TheAgentHamster, 2023-02-24:
[CFFFP_Enterprise_Original.zip](https://github.com/Ultimaker/Cura/files/10819928/CFFFP_Enterprise_Original.zip)

TheAgentHamster, 2023-02-24:
Also, how did you add all the different parts to one project?
Thanks

MariMakes, 2023-02-27:
Hey @TheAgentHamster,

Thanks for sharing the file! 

I was able to recreate your crash, and also saw that if you moved the model slightly away from the center I was still able to slice it. 🎉  I hope that helps you get unstuck. 
[image]

I'll bring your issue up to the team to take a look.

TheAgentHamster, 2023-02-27:
No problem