Ultimaker/Cura #14644: esun PLA black profile

dazzla13, 2023-02-23:
### Application Version

5.3.0

### Platform

mac

### Printer

Ender 3s1 Pro

### Reproduction steps

eSun PLA+ black profile for a 0.04mm nozzle says "Profile not supported"
[image]

### Actual results

I can not print

### Expected results

I used to use this existing profile all the time now with new update it does not work

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

hi

MariMakes, 2023-02-23:
hi. 

Welcome to the Ultimaker Cura Github 🚀 
Thanks for your report 👍

Did you make this PLA+ yourself? 

I recognize your issue, it's quite common.
When making a new material it's important that Material Type matches one of the Generic Materials.
You won't see it anywhere, but Cura needs this material to determine what to base the different layerheight/profiles on.

This seems to be a duplicate of https://github.com/Ultimaker/Cura/issues/8229
We have a ticket on our backlog with the intent to improve this so it's easier to pick a material and the chance that something goes wrong is smaller.

If you did not make this profile, did you mind sharing your configuration file? 
You can share your configuration folder by:
	1. Go to Help at the top bar of Cura
	2. Select Show Configuration Folder
	3. Select all the folders in there and create a zip OR go up a folder in your folder structure and zip your folder with the correct Cura version

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.

ElJefeDSecurIT, 2023-02-24:
i think I might have figured this one out. 
in the picture attached, I have 2 custom materials, and they show the profiles.

[image]

however, when I first created the 2nd one, I accidentally left a space character in the MATERIAL field:

[image]

and then I lose the profiles: 
[image]

this may not be the root issue, but I stumbled across this bug when I saw 2 entries under the same brand: 
[image]

MariMakes, 2023-02-27:
Hey @dazzla13,

Thanks for getting back to us 🙂 

To me, it doesn't seem like a bug in Ultimaker Cura anymore. 
Would you still like to keep this issue open or do you agree that we can close it?

MariMakes, 2023-02-27:
Whoops, I misread. Apologies. 

Good to hear you resolved your issue, @ElJefeDSecurIT.
Did you get a chance to resolve yours, @dazzla13 ?

dazzla13, 2023-02-28:
Hello, My problem is not my own custom profiles it’s just the Esun PLA+ generic one. I can go to the dropdown box on all the others and even the esun PETG  and it works fine. But as soon as I pick the Esun PLA+ black profile it says not supported

[image]<https://user-images.githubusercontent.com/8565295/221063945-2b3f95a7-2c30-4fc0-b700-50635939d339.png>

[image]<https://user-images.githubusercontent.com/8565295/221064155-00bbca4a-2565-4f11-a968-f6563b010273.png>

[image]<https://user-images.githubusercontent.com/8565295/221064237-c0f094d4-6991-4c24-aeb3-3624853b6e90.png>

[image]<https://user-images.githubusercontent.com/8565295/221064326-eecb2d55-b29e-43f2-918d-0c491ca8cd57.png>

dazzla13, 2023-02-28:
Hello No Not yet unfortunately . As I said in my other email My problem is not my own custom profiles it’s just the Esun PLA+ generic one. I can go to the dropdown box on all the others and even the esun PETG  and it works fine. But as soon as I pick the Esun PLA+ black profile it says not supported

This only happened when I did they update.

MariMakes, 2023-03-13:
Hey @dazzla13,

I took a look and there seems to be an error in this PLA+ material. 
[This](https://github.com/Ultimaker/fdm_materials/blob/master/eSUN_PLA_PRO_Black.xml.fdm_material)  <material>PLA+</material> cannot be PLA+, it should be PLA instead

We've added a ticket to the backlog with the intent to improve this. 
For internal reference CURA-10396

Thanks for the report! 👍