Ultimaker/Cura #14647: [5.3.0-alpha+xmas] + [5.2.2] Account Sync -> Cura Crashes

Serious88, 2023-02-23:
### Application Version

5.2.2 / 5.3.0-alpha+xmas

### Platform

Windows 11

### Printer

Creality Ender 3 V2 Neo

### Reproduction steps

1. Start Cura
2. Popup for Account Sync appears
3. Click on "Sync"
4. Cura Freezes

### Actual results

Cura is freezing and gives no answer to the System

### Expected results

I think the settings from my online Account should be syncing

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

Log File:
https://www.file-upload.net/download-15100325/cura.log.html

MariMakes, 2023-02-23:
Hey @Serious88,

Welcome to the Ultimaker Cura Github 🚀 

Your issue reminds me of: https://github.com/Ultimaker/Cura/issues/14328
But it's not completely the same. 

I'll have someone look at your logs 👍 

If you have a crash like this, can you scroll down in the text windows and share a screenshot of what you see?

Serious88, 2023-02-23:
Hey @MariMakes 

if i share you a screenshot, you will just see Cura :D There is nothing that changes/happens after the "Click". The Software just takes no response.