Ultimaker/Cura #14649: Slicer fail with significant difference between wall and infill layer height.

HarryPorpise, 2023-02-23:
### Application Version

Tested on 5.3.0 Xmas & 5.3.0-beta 1

### Platform

Windows 10

### Printer

Creality Ender 3

### Reproduction steps

1. Set layer height to .04mm
2. Set infill layer height to .12mm or above
3. Attempt to slice

### Actual results

Cura will fail to slice the model after an inconsistent amount of time.

### Expected results

Cura should finish slicing the model, no matter how long it takes.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[TIE Fighter Body Front.zip](https://github.com/Ultimaker/Cura/files/10814156/TIE.Fighter.Body.Front.zip)

MariMakes, 2023-02-24:
Hey @HarryPorpise,

Welcome to the Ultimaker Cura Github 🚀 
Thanks for your report 👍

Wow, you changed A LOT of settings. I indeed think that it is related to your infill height setting, but perhaps in combination with your 100% infill. 

Infill Height is often changed to shorten printing time, are you trying to reduce the printing time? 
Because than you can also choose a less high infill percentage or even lightning infill. Especially if the model is not going to be under stress. 

To me, it doesn't seem like a bug in Ultimaker Cura but a combination of incompatible settings. 
Would you still like to keep this issue open or do you agree that we can close it?

HarryPorpise, 2023-02-24:
The issue happens with any infill density or style, and I'm using a high infill density because I want the part to feel a little heavier and higher quality. I have noticed that it will slice in the newest beta if I set "Extra Infill Wall Count" to 1 or above. I think it's still a bug, just not a very high priority one. Maybe just leave it on the backburner so that someone can fix it if they want to for some reason?

jellespijker, 2024-01-14:
Thank you for your report. It seems that your uploaded project file slices correct on our latest unreleased 5.7.0-alpha.0+3abb5 which indicates that the issue you're experiencing seems to be fixed for the upcoming 5.7.0 release.
I'm closing this issue as fixed for the 5.7.0 release
[image]

closed 2024-01-14 by jellespijker