Ultimaker/Cura #14650: Solidworks Macro

darrenseating, 2023-02-23:
### Application Version

4.12.0

### Platform

Windows 10

### Printer

Ultimaker S5

### Reproduction steps

Macro buttin in solidworks
Cura opens up
fails to import solidworks file

### Actual results

fails to import solidworks file 

### Expected results

impost the file and be able to process it 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

MariMakes, 2023-02-24:
Hey @darrenseating,

Welcome to the Ultimaker Cura Github 🚀 
I'm not familiar with this Solidworks Macro, 

I'm not sure if it's an issue in Cura. Solidworks is not a supported file type by default. 
We mainly accept .STL and .3mf files. 

Can you perhaps make a screen recording, or give some more context?
Do you have certain plug-ins installed to make it work? Which plug-ins are those?

darrenseating, 2023-02-28:
Hello,
This is a set up with-in Cura 4.12.0

As you can see below, it has instructions to follow to set it up.

Extensions-3DS SolidWorks Plugin

***@***.***

CAUTION: External Email!

MariMakes, 2023-03-03:
Hey @darrenseating,

Did you mean to include pictures or an attachment? Because I just see:
> As you can see below, it has instructions to follow to set it up.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.

MariMakes, 2023-06-05:
Hey @darrenseating,

We are cleaning up our open issues, so we can focus on the most requested and needed features and bug fixes from our community, and from UltiMaker, I stumbled upon your issue about the Solidworks Macro not working for you. 

To me, it doesn't seem like a bug in Ultimaker Cura.
Would you still like to keep this issue open or do you agree that we can close it?

closed 2023-06-20 by github-actions[bot]