Ultimaker/Cura #14654: Slicing does not work on a model

Djavaherpour, 2023-02-23:
### Application Version

5.2.2

### Platform

Tried on both Mac OS 3.0.1 and Windows 10

### Printer

Ultimaker S3

### Reproduction steps

The slicer cannot finish the process on this mode, even though it is not complicated, it is water-tight, and there are no errors with it. 

### Actual results

There is no response from the slicer even after 2 hours of waiting time. 

### Expected results

The slicer should be able to slice this model in 1-2 minutes max (Simplify 3D slices it in less than 1 minute). 

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[Spheres (Portion) - 2023-02-23.stl.zip](https://github.com/Ultimaker/Cura/files/10816296/Spheres.Portion.-.2023-02-23.stl.zip)

MariMakes, 2023-02-24:
Hey @Djavaherpour,

Welcome to the Ultimaker Cura Github 🚀 
Thanks for your report 👍
Thanks for reaching out! 

[image]
Are you sure you shared the correct model? 
This is a really tiny model 🐜 

Do you have a project file for us? It contains the printer and settings we need for troubleshooting. 
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.

Djavaherpour, 2023-02-24:
Hey @MariMakes 

Thank you so much for reaching out! I totally forgot to mention that the Stl file needs to be scaled to 1000% (10 times bigger. I have attached the project file for your reference. Thank you so much in advance for your advice.
[Spheres_Scaled_for_Cura.3mf.zip](https://github.com/Ultimaker/Cura/files/10826580/Spheres_Scaled_for_Cura.3mf.zip)

MariMakes, 2023-02-27:
Heyyyy @Djavaherpour,

I think I know what is up 👍 

This seems to be a duplicate of https://github.com/Ultimaker/Cura/issues/14108
We are working on a solution as we speak. 
If you set 'Skin Edge Support Layers' to 0, it will slice quicker again!
 
[image]

I'll be closing this issue as a duplicate but you can follow the progress here: https://github.com/Ultimaker/Cura/issues/14108

closed 2023-02-27 by MariMakes